jsk-enshu/robot-programming #347: ubuntuが起動しなくなった

lie-far-away, 2020-10-19:
学科pcのubuntuがフリーズしてしまったため、強制的に電源を切ったあと起動しようとすると、パスワード入力の画面までは表示されるのですがパスワードを入力すると、紫色一色の画面にマウスポインタが表示されるのみでなにもできない状態になってしまいました。windowsの方は問題なく起動するため、ubuntuのguiがうまく起動していないのだと思います。

ちなみにフリーズしたのは学科pcを外部モニターにつないで作業していて、下のリンクの記事を参考にしてfcitxを用いてワイヤレスのusキーボードを導入している最中でした。（どこが関係しているかわかりませんが）
[https://qiita.com/AnchorBlues/items/beeb40593fd4c579dbe5](url)

k-okada, 2020-10-19:
最初にOSを選ぶ画面があるとおもいますが、そこで上から二番目を選んでください。

https://www.maketecheasier.com/boot-recovery-mode-ubuntu/
でrootを選択。
1) qiita みて作業した内容を取り消す。'apt-get remove... 'する
2) 他にいくつか復帰できるかもしれない方法はあるので、連絡ください。

lie-far-away, 2020-10-19:
rootでapt-get remove fcitx fcitx-mozcとしてみましたが未だ起動しません。

k-okada, 2020-10-19:
$su mech-user
でmechユーザになれるとおもうけど，そこで
$ls -al
$cat .xinputrc
$ls -al .local/share/xorg
$cat  .local/share/xorg/Xorg.0.log
$ls -al /etc/X11

家，近ければ学校に来るのが早いかも

lie-far-away, 2020-10-19:
対応していただきありがとうございました。

closed 2020-10-19 by lie-far-away