jsk-enshu/robot-programming #349: our-memberについて

ummaaa, 2020-10-20:
10/19資料10ページのour-member関数はどのように使うのでしょうか。また、atomの意味も教えていただきたいです。

ishiguroJSK, 2020-10-20:
```
$ irteusgl
irteusgl$ (defun our-member (obj lst) (if (atom lst) nil (if (eql (car lst) obj) lst (our-member obj (cdr lst)))))
our-member
irteusgl$ our-member 1.2 (list 1.0 1.1 1.2 1.3)
(1.2 1.3)
irteusgl$ our-member 1.23 (list 1.0 1.1 1.2 1.3)
nil
```
our-memberはリストの中から一致するものを探す関数のようですね．(見つかったらその時点での残りのリストを返す)
eqlで比較しているので，数値やポインタの指すアドレスレベルで一致している必要があります．

> eq obj1 obj2
> obj1 と obj2 が同じオブジェクトを指すポインタあるいは同じ数値のとき T を返す。例えば :(eq ’a ’a)
> は T 、 (eq 1 1) は T 、 (eq 1. 1.0) は NIL 、 (eq "a" "a") は NIL である。

atomはリストのように分割できるものか，数値のように分割できないものかを判定する関数で（本当は他にもいろいろあるのですが※jmanual参照↓）

> atom object
> オブジェクトが cons のインスタンスである時のみ、 NIL を返す。 (atom nil) = (atom ’()) = T)
> 注意 :vectors, strings, read-table, hash-table などに対しては、それらがどんなに複雑なオブジェクトと
> なっていても atom は T を返す。

ここではリストが空かどうかの判定に使用しています

> month-lengthを資料そのまま写したのですが、'feb以外、例えば(month-length 'mar)などとするとnilとなってしまいます。

これは確かにそのようですね．資料の間違いでしょうか？@k-okada

```
irteusgl$ (defun month-length (mon) (case mon (('jan 'mar 'may 'jul 'aug 'oct 'dec) 31) (('apr 'jun 'sept 'nov) 30) ('feb 28)))
month-length
irteusgl$ month-length 'feb
28
irteusgl$ month-length 'jan
nil
```

ummaaa, 2020-10-21:
丁寧なご回答ありがとうございました。

k-okada, 2020-10-22:
一瞬バグかな？とおもってhttps://github.com/euslisp/EusLisp に Issue 書きかけたんですが

```
1.eus0$ (case :b ((:a :b :c) 10) (:d 20) (:e 20))
10
2.eus0$ (case :d ((:a :b :c) 10) (:d 20) (:e 20))
20
3.eus0$ (case 'b (('a 'b 'c) 10) ('d 20) ('e 20))
nil
4.eus0$ (case 'd (('a 'b 'c) 10) ('d 20) ('e 20))
20
5.eus0$ (case 'b ((a b c) 10) (d 20) (e 20))
10
6.eus0$ (case 'd ((a b c) 10) (d 20) (e 20))
20
```
ということで，資料の間違いでした．すいません．
https://euslisp.readthedocs.io/en/latest/controls.html?highlight=case#conditionals
```
case key &rest (label &rest forms)

For the clause whose label matches with key, forms are evaluated and the last value is returned. Equality between key and label is tested with eq or memq, not with equal.
```
とあるんですがこのlabelの部分が既にquote されているので以下のように評価されているということですかね．
```
7.eus0$ (memq 'b '(a b c))
(b c)
8.eus0$ (memq 'b '('a 'b 'c))
nil
9.eus0$ (memq :b '(:a :b :c))
(:b :c)
```

ummaaa, 2020-10-23:
ありがとうございました。

closed 2020-10-23 by ummaaa