jsk-enshu/robot-programming #351: 反復オペレータdoについて

kohtaro246, 2020-10-23:
教材に載っているshow-squaresやhttp://www.nct9.ne.jp/m_hiroi/clisp/abcl05.htmlにのっているdoをつかったプログラムをemacs lispで動かそうとすると、写真のようなエラーが出ます。原因をご教示いただけると幸いです。
[image]

ishiguroJSK, 2020-10-23:
全然詳しくないんだけど，emacs lispってcommon lispと結構違うという認識で，common lispとかeuslispのコードはいろいろ動かないとは思います．

kohtaro246, 2020-10-23:
あ、そうなんですか！！
てっきりlispだったらなんでも同じかと思ってました。

k-okada, 2020-10-26:
emacs lisp だと
```
(cl-do ((i 0 (+ i 1)))
    ((> i 10) 'done)
  (print i))
```
ですね．

プログラム言語の仕様と実装は異なるモノ，というのが結構一般的で
C言語なんかもGCCとMicroSoftのもので微妙に違ったりします．
例：https://easyaspi314.github.io/gcc-vs-clang.html

kohtaro246, 2020-10-28:
ありがとうございます。

closed 2020-10-28 by kohtaro246