jsk-enshu/robot-programming #352: inconsistent face settingerr:

kohtaro246, 2020-10-28:
このエラーはなぜ起きるのでしょうか。
コード：https://gist.github.com/kohtaro246/b7fad841b134d2904e9770155f2ab10b
エラー画面
[image]

k-okada, 2020-10-29:
遅くなりました．
面と面が重なっているところの`body-`は上手く行かないことが多いです．
```
    ;;make-handle                                                                                                       
    (setq b4 (make-cube 60 10 3))
    (setq b3 (make-cube 54 6 5))
    (send b4 :locate #f(30 -5 38.5))
    (send b3 :locate #f(30 -3.5 38.5))
    (setq b4 (body- b4 b3))
    (setq b1 (body+ b1 b4))
```
などと，少し値を変えてあげると上手く動きます．

kohtaro246, 2020-10-29:
解決しました。ご返信ありがとうございます。

closed 2020-10-29 by kohtaro246