jsk-enshu/robot-programming #353: (body+ (b1 b2))が通らない

softyanija, 2020-10-29:
モデリングをしているときに、次のようなコードを書いてオブジェクトを作ろうとするとエラーになります。調べてみると面と面が接しているとこのようなバグを吐かれるらしいですが、今回はそのような配置になっていないはずです。どうすればいいでしょうか？
'''
(defun make-gun nil
  (let (b1 b2 b3)
    (setq b1 (make-cylinder 2 40))
    (send b1 :rotate (deg2rad 90) :y)
    (send b1 :set-color :blue)
    (setq b2 (make-cylinder 5 10))
    (send b2 :rotate (deg2rad 90) :x)
    (send b2 :locate #f(0 0 -5))
    (setq b3 (body+ b1 b2))
    b1))

irteusgl 0 error: outer circuit not found
 in (body+ b1 b2)
'''
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

k-okada, 2020-10-29:
```
(defun make-gun nil
  (let ();b1 b2 b3)                                                                                                     
    (setq b1 (make-cylinder 20 400))
    (send b1 :rotate (deg2rad 90) :y)
    (send b1 :set-color :blue)
    (setq b2 (make-cylinder 50 100))
    (send b2 :rotate (deg2rad 90) :x)
    (send b2 :rotate (deg2rad 45) :z)
    (send b2 :locate #f(0 0 -50))
    (objects (list b1 b2))
    (setq b3 (body+ b1 b2))
    b3))

```
とすると動きました．
単位は`mm`で，等身大ロボットターゲットメインのシステムなので，余り小さいものはしきい値が対応していないので，そこそこ大きいものを作ったほうが良いのと，
上手くCylinderのエッジが重なるようにしたら上手くいきました．

ただ，こいうときは`bodyset`使うと良いです．

softyanija, 2020-10-29:
エラーが無くなりました。ありがとうございます。

closed 2020-10-29 by softyanija