jsk-enshu/robot-programming #354: アニメーションが実行されない

gawa-K, 2020-10-29:
資料のプログラムの一部を変更してサーボ１つを動かそうとしています。サーボが２つの場合はアニメーションできていたのですが、変更したとたんに`(load "ファイル名")`で`t`が返ってくるだけで表示されなくなりました。どこがおかしいのでしょうか？
以下が実行しようとしているプログラムです。
```
(defun make-servo nil
(let (b1 b2)
(setq b1 (make-cube 35 20 46))
(send b1 :locate #f(9.5 0 0))
(setq b2 (make-cylinder 3 60))
(send b2 :locate #f(0 0 -30))
(setq b1 (body+ b1 b2))
(send b1 :set-color :gray20)
b1)) 

(defun make-hinji nil
(let ((b2 (make-cube 22 16 58))
(b1 (make-cube 26 20 54)))
(send b2 :locate #f(-4 0 0))
(setq b2 (body- b2 b1))
(send b1 :set-color :gray80)
b2))
(setq h1 (instance bodyset-link :init (make-cascoords) :bodies (list (make-hinji))))
(setq s1 (instance bodyset-link :init (make-cascoords) :bodies (list (make-servo))))
(setq j1 (instance rotational-joint :init :parent-link h1 :child-link s1 :axis :z))

(setq r (instance cascaded-link :init))
(send r :assoc h1)
(send h1 :assoc s1)
(setq (r . links) (list h1 s1))
(setq (r . joint-list) (list j1))
(send r :init-ending) 

(defclass hinji-servo-robot
:super cascaded-link)
(defmethod hinji-servo-robot
(:init
()
(let (h1 s1 l1 l2)
(send-super :init)
(setq h1 (make-hinji))
(setq s1 (make-servo))
(setq l1 (instance bodyset-link :init (make-cascoords) :bodies (list h1)))
(setq l2 (instance bodyset-link :init (make-cascoords) :bodies (list s1)))
(send l2 :locate #f(21 0 0))
(send self :assoc l1)
(send l1 :assoc l2)
(setq joint-list
(list (instance rotational-joint
:init :parent-link l1 :child-link l2
:axis :z)
(setq links (list l1 l2))
(send self :init-ending)
)))
(setq r (instance hinji-servo-robot :init))
(objects (list r))
(dotimes (i 10000)
(send r :angle-vector (float-vector (* 90 (sin (/ i 500.0))))) 
(send *irtviewer* :draw-objects)) 
```

gawa-K, 2020-10-30:
他の方法を見つけて解決しました。

closed 2020-10-30 by gawa-K

k-okada, 2020-10-30:
遅くなりました．どういう方法で解決したかな？

`hinji-servo-robot` の部分でカッコの対応が取れていなかったようです．
コピーペースト失敗だけかもしれないけど，`Esc-<`, `Ctrl-space`, `Esc->` でファイルを全部選択して`Esc-x indent-region` でインデントを整理すると間違いが見つけやすいです．

```
(defclass hinji-servo-robot
  :super cascaded-link)
(defmethod hinji-servo-robot
  (:init
   ()
   (let (h1 s1 l1 l2)
     (send-super :init)
     (setq h1 (make-hinji))
     (setq s1 (make-servo))
     (setq l1 (instance bodyset-link :init (make-cascoords) :bodies (list h1)))
     (setq l2 (instance bodyset-link :init (make-cascoords) :bodies (list s1)))
     (send l2 :locate #f(21 0 0))
     (send self :assoc l1)
     (send l1 :assoc l2)
     (setq joint-list
           (list (instance rotational-joint
                           :init :parent-link l1 :child-link l2
                           :axis :z)))
     (setq links (list l1 l2))
     (send self :init-ending)
     self))
  )
```

gawa-K, 2020-10-30:
御返事ありがとうございます。
プログラムを書き直したらうまくいきました。
ただ先生のおっしゃる通り括弧の対応がうまくいっておらず、アニメーションが表示されず`t`のみ返ってくることがありました。
ご親切なご対応ありがとうございます。