jsk-enshu/robot-programming #355: make-cubeでのエラー

softyanija, 2020-10-31:
下のコードを実行すると時折make-cubeの部分でエラーを吐かれます。
~~~
(defun make-body nil
  (let (b1 b2)
    (setq b1 (make-cube 500 400 150))
    (send b1 :locate #f(0 0 75))
    (setq b2 (make-cylinder 50 10)) 
    (send b2 :locate #f(0 0 140))
    (setq b1 (body- b1 b2))
    (send b1 :set-color :red)
    b1))

(defun make-gear nil
  (let (b1)
    (setq b1 (make-cylinder 50 10))
    (send b1 :locate #f(0 0 -10))
    (send b1 :set-color :gray20)
    b1))

(defun make-turret nil
  (let (b1)
    (setq b1 (make-cube 200 200 100))
    (send b1 :locate #f(0 0 50))
    (send b1 :set-color :gray20)
    b1))

(defun make-defensive nil
  (let (b1)
    (setq b1 (make-cylinder 25 50))
    (send b1 :rotate (deg2rad 90) :x)
    (send b1 :locate #f(0 0 -25))
    (send b1 :set-color :gray20)
    b1))

(defun make-gun nil
  (let (b1)
    (setq b1 (make-cylinder 10 200))
    (send b1 :rotate (deg2rad 90) :y)
    (send b1 :set-color :blue)
    b1))

;(setq t1 (make-gun))
;(objects (list t1))

(defclass tiger
  :super cascaded-link)
(defmethod tiger
  (:init
   ()
   (let (body gear turret defensive gun base head cannon)
     (send-super :init)
     (setq body (make-body))
     (setq gear (make-gear))
     (setq turret (make-turret))
     (setq defensive (make-defensive))
     (setq gun (make-gun))
     (send gear :locate #f(0 0 150))
     (send turret :locate #f(0 0 150))
     (send defensive :locate #f(100 0 200))
     (send cannon :locate #f(100 0 200))
     (send gear :assoc turret)
     (send defensive :assoc gun)
     (setq base (instance bodyset-link :init (make-cascoords) :bodies (list body)))
     (setq head (instance bodyset-link :init (make-cascoords) :bodies (list gear turrete)))
     (setq cannon (instance bodyset-link :init (make-cascoords) :bodies (list defensive gun)))
     (send self :assoc base)
     (send base :assoc head)
     (send head :assoc cannon)
     (setq joint-list
	   (list (instance rotational-joint
			   :init :parent-link base :child-link head
			   :axis :z)
		 (instance rotational-joint
			   :init :parent-link head :child-link cannon
			   :axis :y)))
     (setq links (list base head cannon))
     (send self :init-ending)
     )))
(setq r (instance tiger :init))
(objects (list r))

;(dotimes (i 10000)
;(send r :angle-vector (float-vector (* 90 (sin (/ i 500.0))) (* 90 (sin (/ i 1000.0)))))
(send *irtviewer* :draw-objects))
~~~
~~~
5.E4-irteusgl$ (load "1026-2.l")
Call Stack (max depth: 20):
  0: at (make-cube 500 400 150)
  1: at (setq b1 (make-cube 500 400 150))
  2: at (let (b1 b2) (setq b1 (make-cube 500 400 150)) (send b1 :locate #f(0.0 0.0 75.0)) (setq b2 (make-cylinder 50 10)) (send b2 :locate #f(0.0 0.0 140.0)) (setq b1 (body- b1 b2)) (send b1 :set-color :red) b1)
  3: at (make-body)
  4: at (setq body (make-body))
  5: at (let (body gear turret defensive gun base head cannon) (send-super :init) (setq body (make-body)) (setq gear (make-gear)) (setq turret (make-turret)) (setq defensive (make-defensive)) (setq gun (make-gun)) (send gear :locate #f(0.0 0.0 150.0)) (send turret :locate #f(0.0 0.0 150.0)) (send defensive :locate #f(100.0 0.0 200.0)) (send cannon :locate #f(100.0 0.0 200.0)) (send gear :assoc turret) (send defensive :assoc gun) (setq base (instance bodyset-link :init (make-cascoords) :bodies (list body))) (setq head (instance bodyset-link :init (make-cascoords) :bodies (list gear turrete))) (setq cannon (instance bodyset-link :init (make-cascoords) :bodies (list defensive gun))) (send self :assoc base) (send base :assoc head) (send head :assoc cannon) (setq joint-list (list (instance rotational-joint :init :parent-link base :child-link head :axis :z) (instance rotational-joint :init :parent-link head :child-link cannon :axis :y))) (setq links (list base head cannon)) (send self :init-ending))
  6: at (send #:inst69 :init)
  7: at (let ((#:inst69 (instantiate tiger))) (send #:inst69 :init) #:inst69)
  8: at (instance tiger :init)
  9: at (setq r (instance tiger :init))
  10: at (load "1026-2.l")
  11: at euserror
  12: at euserror
  13: at (make-cube 10 10 10)
  14: at (setq b1 (make-cube 10 10 10))
  15: at euserror
  16: at euserror
  17: at (make-cube 50 40 15)
  18: at (setq b1 (make-cube 50 40 15))
  19: at euserror
  And more...
irteusgl 0 error: class expected in (make-cube 500 400 150)
~~~
サイズが原因かと思い(10 10 10) (50 40 15)といったサイズで試しても同じようになります。また、irteusglを立ち上げ直すと確率でエラーが出なくなります。これは何が原因なんでしょうか？

k-okada, 2020-10-31:
`body`はすでにクラス名としてシステムに使っているので，
```
-   (let (body gear turret defensive gun base head cannon)  
+   (let (abody gear turret defensive gun base head cannon)
```
としましょう．
https://github.com/euslisp/EusLisp/blob/845d35fbf51c7a8f1f4cfe10024da5478ecde1bd/lisp/geo/geoclasses.l#L100

僕の環境だと，irteusを立ち上げ直すと必ずエラーになるけど．．．
```
5.E4-irteusgl$ (load "1026-2.l")
```
で
`<数字>.-irteusgl$` ではなくて，`<数字>.E<数字>-irteusgl$`となっているときはエラーモードなので，`(reset)`としてやり直すのがよいです．
http://euslisp.github.io/EusLisp/jmanual-node12.html

softyanija, 2020-10-31:
解決しました、ありがとうございます！

closed 2020-10-31 by softyanija