jsk-enshu/robot-programming #356: 時計を動かしたい

lie-far-away, 2020-10-31:
写真のような時計のアニメーションを動かしたいのですが、segmentation faultが出て動きません。
エラーメッセージ
```
1.irteusgl$ (load "animation.l")
;;2 face-to-face contact(s) found.
;;2 face-to-face contact(s) found.
;;1 face-to-face alignment(s) found.
;;2 face-to-face contact(s) found.
;;2 face-to-face contact(s) found.
;;1 face-to-face alignment(s) found.
;; (make-irtviewer) executed
;; Segmentation Fault.
;; in (send r :angle-vector (float-vector (* 1.2 i) (* 6 i)) (send *irtviewer* :draw-objects))
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=1666279664 x=635167c0 addr=b6
Fatal: 

```

コード
```
(defun make-longhand nil
  (let (b1 b2)
    (setq b1 (make-cube 40 600 30))
    (send b1 :locate #f(0 260 120))
    (setq b2 (make-cylinder 20 30))
    (send b2 :locate #f(0 0 105))
    (setq b1 (body- b1 b2))
    (send b1 :rotate (deg2rad 60) :z)
    (send b1 :set-color :purple)
    b1))
(defun make-shorthand nil
  (let (b1 b2)
    (setq b1 (make-cube 40 450 30))
    (send b1 :locate #f(0 185 152))
    (setq b2 (make-cylinder 20 30))
    (send b2 :locate #f(0 0 137))
    (setq b1 (body- b1 b2))
    (send b1 :set-color :green)
    b1))
(defun make-board nil
  (let (b1 b2)
    (setq b1 (make-cylinder 800 100))
    (send b1 :locate #f(0 0 0))
    (setq b2 (make-cylinder 20 200))
    (send b2 :locate #f(0 0 0))
    (setq b1 (body+ b1 b2))
    (send b1 :set-color :white)
    b1))
(setq lh (instance bodyset-link :init (make-cascoords) :bodies (list (make-longhand))))
(setq sh (instance bodyset-link :init (make-cascoords) :bodies (list (make-shorthand))))
(setq bd (instance bodyset-link :init (make-cascoords) :bodies (list (make-board))))
(setq j1 (instance rotational-joint :init :parent-link bd :child-link lh :axis :z))
(setq j2 (instance rotational-joint :init :parent-link bd :child-link sh :axis :z))

;; instance cascaded coords
(setq r (instance cascaded-link :init))
(send r :assoc bd)
(send bd :assoc lh)
(send bd :assoc sh)
(setq (r . links) (list bd lh sh))
(setq (r . joint-list) (list j1 j2))
(send r :init-ending)

(defclass my-clock
  :super cascaded-link)
(defmethod my-clock
  (:init
   ()
   (let (lh sh bd l1 l2 l3)
     (send-super :init)
     (setq lh (make-longhand))
     (setq sh (make-shorthand))
     (setq bd (make-board))
     (setq l1 (instance bodyset-link :init (make-cascoords) :bodies (list lh)))
     (setq l2 (instance bodyset-link :init (make-cascoords) :bodies (list sh)))
     (setq l3 (instance bodyset-link :init (make-cascoords) :bodies (list bd)))
     (send l3 :locate #f(0 0 0))
     (send self :assoc l3)
     (send l3 :assoc l1)
     (send l3 :assoc l2)
     (setq joint-list
	   (list (instance rotational-joint
			   :init :parent-link l3 :child-link l2
			   :axis :z)
		 (instance rotational-joint
			   :init :parent-link l3 :child-link l1
			   :axis :z)))
     (setq links (list l1 l2 l3))
     (send self :init-ending)
     ))
  )
(setq r (instance my-clock :init))
	      
(objects (list r))

(dotimes (i 300)
  (send r :angle-vector (float-vector (* 1.2 i) (* 6 i))
  (send *irtviewer* :draw-objects))
```
[image]

k-okada, 2020-11-01:
お，いいですね．
```
(dotimes (i 300)
  (send r :angle-vector (float-vector (* 1.2 i) (* 6 i)))
  (send *irtviewer* :draw-objects))

```
とカッコが一つ抜けていますね．あと，
```
     (setq joint-list
           (list (instance rotational-joint
                           :init :parent-link l3 :child-link l2
                           :min *-inf* :max *inf*
                           :axis :z)
                 (instance rotational-joint
                           :init :parent-link l3 :child-link l1
                           :min *-inf* :max *inf*
                           :axis :z)))
```
とすると無限回転になります．

lie-far-away, 2020-11-01:
動かしてみたら反時計回りだったので角度を負にして完成しました。ありがとうございました。

closed 2020-11-01 by lie-far-away