jsk-enshu/robot-programming #358: 10/26の資料p8のservo-modelクラスについて

ketaro-m, 2020-11-01:
10/26の資料p8に下のようにあり、`rtmodel.l で実装されているクラスの概要．打ち込まないこと．`とあるのですが、これはcascaded-linkクラスは打ち込まず、servo-modelクラスは自分で書き込めということでしょうか？
初めに何も打ち込まずに、`(setq r (instance servo-model :init))`とだけするとエラーが出たので、[https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l](https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l)を覗いてみたところ、servo-modelクラスは存在していなかったので、このクラスだけ作成し、さらにその中で使われているmake-servoやmake-hinjiといった関数もp6~7にかけて存在しているので、その通りに打ち込むとオブジェクトを作ることができました。
このような解釈であっていますか？

また、もしそうだとしたら、資料の文言が若干分かりにくいような気がするので、上記のことを反映していただけると来年以降の僕のように混乱する学生が減るのではないかと思うのですが、いかがでしょうか。

[image]

k-okada, 2020-11-01:
はい，その解釈であっています．

７ページに書いてある方法で，ベタ書きでロボットモデルは生成出来ているはずなので，
これをそのままクラス定義の中に入れる，という作業をすることを期待しています．
脂漏は直しておきます．
--
◉ Kei Okada

ketaro-m, 2020-11-01:
お返事ありがとうございます。

closed 2020-11-01 by ketaro-m