jsk-enshu/robot-programming #360: 11/02, 宿題１:  Client [/cmd_vel_renamer] wants topic /cmd_vel to have datatype/md5sum.....

Masayoshi-Dohi, 2020-11-02:
宿題１で以下の写真のようなエラーが出ました。
メッセージの型が合っていないようなのですが、調べても解決法が分かりませんでした。
このエラーの解決法に加えてデバッグの方法などについても教えていただけると幸いです。
よろしくお願いいたします。

[image]

k-okada, 2020-11-03:
おかしいですね．デバッグの方法ですが，先ずは問題を再現する最小のプログラムを作るのが良いです．僕は
```
#!/usr/bin/env roseus

(ros::roseus-add-msgs "geometry_msgs")
(ros::roseus "cmd_vel_pub")
(ros::advertise "/cmd_vel" geometry_msgs::Twist 5)

(ros::rate 10)
(do-until-key
 (setq msg (instance geometry_msgs::Twist :init))
 (send msg :linear :x 0.1)
 (ros::publish "/cmd_vel" msg)
 (ros::ros-info "~A" msg)
 (ros::sleep)
 (ros::spin-once))
```
を作ってみました．

このプログラムのポイントは「動いているハズ」と思っているところを確実に確認する事が重要ですね．万が一ですが，最初に１回エラーが出て，後は動いている，ということもあるかもしれないので．
ということで，を`ros::publish`の前後に
```
 (ros::ros-info "~A" cmd_vel)
```
と入れて，確実にpublish されているか確認します．僕の場合はこれで動きました．
これでダメなら，もう一回gazeboのプログラムを起ち上げなおして，ちゃんとトピックを受け取るプログラムがいるか調べるために

```
$ rostopic info /cmd_vel
```
とします．この先はちょっと今回の特殊ケースになるんだけど，
```
$ rostopic info /cmd_vel
Type: unknown type

Publishers: None

Subscribers: 
 * /cmd_vel_renamer (http://p51s:41481/)

```
となります．普通は`Type: `に`geometry_msgs/Twist`があることが期待されます．
で，おかしいな，と持って，
```
$ ps -auxww | grep cmd_vel_ren
k-okada  14761  1.0  0.0 338376 10512 ?        Ssl  09:29   0:05 /opt/ros/melodic/lib/topic_tools/relay /cmd_vel /mobile_base/commands/velocity __name:=cmd_vel_renamer __log:=/home/k-okada/.ros/log/90ffb7ba-1d6b-11eb-95e7-00bb605aa0af/cmd_vel_renamer-3.log
k-okada  15914  0.0  0.0  15460  1112 pts/1    S+   09:37   0:00 grep --color=auto cmd_vel_ren
```
として，これがhttp://wiki.ros.org/topic_tools/relay だとわかる．確かに
```
$ rostopic  info /mobile_base/commands/velocity
```
とすると，`Type: geometry_msgs/Twist` になっていることがわかる．relay ははいってきたメッセージの名前を変えるだけだから，まぁこういうものかと考えて一回調査終了．

次の作戦は必ず動くプログラムで確認．
```
$ rosrun teleop_twist_keyboard teleop_twist_keyboard.py 
```
でキーボードで動くかどうか確認．僕はこれで動く．で確かに，
```
$ rostopic info /cmd_vel
Type: geometry_msgs/Twist
```
とTypeも正しく治っている．

で，僕はTye mismatch のエラーがではいないけど，
```
$ rosmsg md5 geometry_msgs/Twist
9f195f881246fdfa2798d1d3eebca84a
```
としてみる．エラーメッセージと見比べると，client側のメッセージの方が違っていそう．
```
$ rosversion topic_tools
1.14.10
$ rosversion geometry_msgs
1.12.7
```
として，バージョンを確認．もし手元のパッケージのバージョンがこれより古ければ
```
$ sudo apt install ros-melodic-topic-tools
```
などとしてインストール
https://github.com/ros/ros_comm/blob/melodic-devel/tools/topic_tools/CHANGELOG.rst
https://github.com/ros/common_msgs/blob/jade-devel/geometry_msgs/CHANGELOG.rst
ただ，これを見ても余りわからないことが多い．あえて言うと，topic_toolあたりが最近新しくなったのが原因か．

で，結局良くわからないんだけど，ライブラリ間のバージョンの不整合が起きていると仮定すると
```
$ sudo apt update && sudo apt dist-upgrade
```
とするとよい．ただ，これでROS以外のプログラムもアップデートが掛かって不具合が起こることもあるので，要注意．インストール直前に`[Y/n]`と聞かれるところで，何がインストールされているか確認して，知らないモノや基礎的なライブラリまでアップデーていそうだったら，まずは以下をやって見る．
```
$ sudo apt updat
$ dpkg --get-selections | grep ros-melodic | cut -f 1 -d$'\t' | xargs sudo apt install --reinstall -y
```
という感じでしょうか？

先ずは`teleop_twist_keyboard.py`で動くかどうかというところで，自分のプログラムの問題か，システムの問題か切り分けができそうです．

Masayoshi-Dohi, 2020-11-03:
丁寧なご回答ありがとうございます。
エラー文をもう一度表示させるためにとりあえず昨日と全く同じ状況で実行してみたところ、なぜか期待通りに動きました。
先生のご説明だと昨日は認識されていなかったと思われるcmd_velのTypeが今日はなぜか認識されているということになると思うのですが、これにはどのような原因が考えられますか？

k-okada, 2020-11-03:
apt-get が走って偶然上手くダウンロードされたか，（ログイン直後にバックグラウンドで走る場合がある）
あとは，workspaceのsourceが機能とは違う状況だったか？ということでしょうか？
今度おかしかったらrosveresion など実行してみるとよいかと思います．

Masayoshi-Dohi, 2020-11-03:
sourceは.bashrcに書き込んでいるので同じ状況だと思います。偶然ということがあるのですね。
ありがとうございました。

closed 2020-11-03 by Masayoshi-Dohi