jsk-enshu/robot-programming #362: 11/2 課題1について

ham-sh, 2020-11-03:
10/12 課題3で書いた以下のプログラムを見ながらroseusに書き換えていますが、エラーが検出され上手く動きませんでした。
正直なところ、課題の意味をいま一つ理解できていないような気もします。
仮にこのプログラムが通ったとして、11/2課題1の要求をクリアしているのでしょうか。
以上二点の質問にお答えしていただけると幸いです。

<10/12 課題3のプログラム (python)>
import rospy
from opencv_apps.msg import RotatedRectStamped
from image_view2.msg import ImageMarker2
from geometry_msgs.msg import Point
from geometry_msgs.msg import Twist

rect = RotatedRectStamped()
def cb(msg):
    global rect
    rect = msg

if __name__ == '__main__':
    try:
        rospy.init_node('client')
        rospy.Subscriber('/camshift/track_box', RotatedRectStamped, cb)
        pub = rospy.Publisher('/cmd_vel', Twist)
        rate = rospy.Rate(10)
        while not rospy.is_shutdown():
            cmd_vel = Twist()
            if (rect.rect.center.x>0 and rect.rect.center.x<300):
                cmd_vel.angular.z = 0.1
                pub.publish(cmd_vel)
                rate.sleep()
            elif (rect.rect.center.x>340 and rect.rect.center.x<640):
                cmd_vel.angular.z = -0.1
                pub.publish(cmd_vel)
                rate.sleep()
            elif (rect.rect.center.x==0):
                cmd_vel.angular.z = 0.2
                pub.publish(cmd_vel)
                rate.sleep()

    except rospy.ROSInterruptException: pass

<11/02 課題1のプログラム(roseus)>
#!/usr/bin/env roseus

(ros::roseus-add-msgs "geometry_msgs")
(ros::roseus-add-msgs "oppencv_apps")                              

(defun cb (msg)
  (setq rect msg))

(ros::rate 10)
(do-until-key
 (ros::roseus "client" :anonymous t)
 (ros::advertise "/cmd_vel" geometry_msgs::Twist 5)
 (setq cmd_vel (instance geometry_msgs::Twist :init))
 (ros::subscribe "/camshift/track_box" opencv_apps::RotatedRectStamped #'cb)                                                  
 (ros::publish "/cmd_vel" cmd_vel)
 (ros::sleep)
 (ros::spin-once))

<roseusのプログラムで検出されたエラー>
configuring by "/opt/ros/melodic/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; extnum ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph ___time ___pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtstl irtwrl irtpointcloud eusbullet bullet irtcollision irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x55b928b82690[16374] --> 0x55b929008230[32748] top=3c74
irtgl irtglc irtviewer 
EusLisp 9.27( 1.2.2) for Linux64 created on ip-172-30-1-193(Wed Aug 12 20:03:47 UTC 2020)
roseus ;; loading roseus("1.7.4") on euslisp((9.27 ip-172-30-1-193 Wed Aug 12 20:03:47 UTC 2020  1.2.2))
eustf roseus_c_util [ERROR] [1604372136.795658543]: Could not find roseus messages for oppencv_apps under (/home/mech-user/catkin_ws/devel/share /opt/ros/melodic/share)
try rosrun roseus generate-all-msg-srv.sh oppencv_apps
;; can not find nil/msg/ directory for [oppencv_apps] packages
[ INFO] [1604372136.795774097]: cell* ROSEUS_EXIT(context*, int, cell**)

k-okada, 2020-11-03:
`oppencv_apps` -> `opencv_apps` かな

課題としては半分OKです．
```
 (setq cmd_vel (instance geometry_msgs::Twist :init))
 (ros::publish "/cmd_vel" cmd_vel)
```
となっていてロボットが動かないと思うので，
`/camshift/track_box`
の値に応じて，ロボットが右回転したり，左回転したり，前後に動くようになっていると良いとおもいます．

あと，`do-until-key` はPythonの`while not rospy.is_shutdown():` に対応しているループ部分ですが，
ループ毎に必要な処理と，プログラムを実行したら一回だけ実行したら良い処理（例えば` (ros::roseus "client" :anonymous t)`等）と
考えて，１回だけ実行したらよい処理はループの外に出すのが良いですね．

ham-sh, 2020-11-03:
なるほど！
ありがとうございます。

closed 2020-11-03 by ham-sh