jsk-enshu/robot-programming #363: 11/02, 宿題2: move_baseサーバーの立ち上げ方

Masayoshi-Dohi, 2020-11-03:
宿題２ではMoveBaseのサーバーも自作する必要があるのでしょうか？

[image]

k-okada, 2020-11-03:
ごめんなさい．

```$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_navigation.launch```と立ち上げる必要があります．

これ知らないと課題できないので，３年生のスラックか何かで皆さんに伝えてもらえればとおもいます．このとき

```Warning: Invalid argument "/camera_depth_frame" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
[ERROR] [1604380038.227598671, 506.870000000]: TF Exception that should never happen for sensor frame: , cloud frame: /camera_depth_frame, Invalid argument "/camera_depth_frame" passed to lookupTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:

 ```

と表示されますが，無視して構わないです．あるいは，https://patch-diff.githubusercontent.com/raw/jsk-enshu/robot-programming/pull/365.diff のように変更してください．

Masayoshi-Dohi, 2020-11-03:
了解しました。
ありがとうございます。

closed 2020-11-03 by Masayoshi-Dohi