jsk-enshu/robot-programming #366: 11/02, 宿題2: ERROR th=0 cannot find method :sequence:sequenc

Masayoshi-Dohi, 2020-11-03:
このエラーが解決できません。Googleで検索しても一切の情報が出てこず、自分で解決するのは厳しいと感じました。
何かわかることがあれば教えていただきたいです。

[image]

k-okada, 2020-11-03:
エラーがでている
`(send (send *c* :get-result) :sequnce)`
という行をコメントアウトして，一回プログラムを動かした後のインタプリタで
`(print *c*)` とか `(print (send *c* :get-result))` としてみてください．
どんなメソッドがあるかは `(send (send *c* :get-result) :methods)`として調べます．

ヒントとしては`(send *c* :get-result)` はROSのメッセージが返ってくるので，それのメンバ変数を調べる必要があります．
このプログラムはfibonacciのプログラムから作っているんだと思いますが，ここの`:sequence`は
https://github.com/ros/common_tutorials/blob/noetic-devel/actionlib_tutorials/action/Fibonacci.action#L5
に対応しているので，同じようにこのプログラムで取得できるresultのなかで知りたい情報にアクセスすることになります．

Masayoshi-Dohi, 2020-11-03:
インタプリタで

(send (send *c* :get-result) :methods)

とすると以下のようなメソッドが表示されました。

(:init :serialization-length :serialize :deserialize :init :md5sum- :datatype- :connection-header :plist :get :put :name :remprop :prin1 :prin1 :warning :error :slots :methods :super :get-val :set-val)

しかしこれらのメソッドを一通り実行してみてもそれらしき値は表示されず、MoveBase.actionのようなファイルがどこかにあるのかと思い検索してみましたが見つかりませんでした。
申し訳ありませんがメンバ変数の調べ方をもう少し詳しく教えていただけませんか？

[image]

ishiguroJSK, 2020-11-03:
これ僕もさっき少し試したんですが，move_baseのgoalは正しく送れていて，resultの表示時にエラーが起こってるだけで，
そもそもMoveBaseActionのresultは何も返さない気がします．
```
$ roscd move_base_msgs/
 /opt/ros/melodic/share/move_base_msgs $ cat action/MoveBase.action 
geometry_msgs/PoseStamped target_pose
---
---
geometry_msgs/PoseStamped base_position

```

Masayoshi-Dohi, 2020-11-03:
なるほど。先程インタプリタで試行錯誤した後にシミュレーターの画面をよく見るとgazeboが少しだけ動いていました。
しかしながらシミュレーターを立ち上げなおしてもう一度実行しようとしてもその挙動を再現できません。
サーバーからresultが返ってきたときに何をすればgazeboは動いてくれるのでしょうか？

Masayoshi-Dohi, 2020-11-03:
目標位置への移動はしてくれませんが、gazeboがその場で回転するのはmove_baseのクライアントを実行した時のみのようです。
一応ぎりぎり課題の要件は満たしている気がしてきました…

ishiguroJSK, 2020-11-03:
テキスト的には，gazeboじゃなくて`roslaunch navigation_stage move_base_amcl_2.5cm.launch`でやる流れだと思うけど，
gazeboでやるなら
```
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_navigation.launch
```
で，

> その場で回転する

はgazebo環境内が狭いから指定したゴールへの経路が見つかってないんじゃないかな？2~3[m]で試したらできると思う．

> サーバーからresultが返ってきたときに何をすれば

result待たずとも動き始めるはずなんだけど，今回はそもそもresult何も返ってこないっぽいから，
```
$ rostopic echo /move_base/result 
header: 
  seq: 0
  stamp: 
    secs: 21
    nsecs: 190000000
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 15
      nsecs: 790000000
    id: "/move_base-1-15.790000000"
  status: 3
  text: "Goal reached."
result: 

---
```
強いていうなら
```
$ send *c* :get-goal-status-text
"Goal reached."
```
とかを表示するのかな・・・（結構難解な気が）

ishiguroJSK, 2020-11-03:
なるほど，岡田先生が既に
```
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_navigation.launch
```
でやるようにアナウンスしているようですね

Masayoshi-Dohi, 2020-11-04:
ゴールの位置を調節したらうまくいきました！
ありがとうございます。

closed 2020-11-04 by Masayoshi-Dohi