jsk-enshu/robot-programming #371: ros::rate, ros::sleepができない

atnarita, 2020-11-05:
課題3について以下のようなコードを書いたのですが(ros::sleep)で停止してしまいます。
whileの前に(ros::rate 1)をおいたので1Hzで回ってほしいのですがどこがおかしいのでしょうか。

```lisp
#! /usr/bin/env roseus

(ros::roseus-add-msgs "trajectory_msgs")

(ros::roseus "send_joint_position" :anonymous t)
(ros::advertise "/fullbody_controller/command" trajectory_msgs::JointTrajectory)

(defun send_joint_position nil
  (ros::ros-info "send_joint_position function !")

  (setq i 0)
  (ros::rate 1)
  (while (ros::ok)
    (progn
      (setq joint_trajectory (instance trajectory_msgs::JointTrajectory :init))
      (send joint_trajectory :header :stamp (ros::time-now))
      (send joint_trajectory :joint_names
            '("arm_joint1" "arm_joint2" "arm_joint3" "arm_joint4" "arm_joint5" "arm_joint6" ))
                                                                    
      (setq point (instance trajectory_msgs::JointTrajectoryPoint :init))
      (send point :positions '((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2)))
      (ros::ros-info "joint [~A]" (send point :positions))
                                                  
      (send point :time_from_start (instance ros::Duration :init 1))                                                                                    
      (send joint_trajectory :points)

      (ros::publish "/fullbody_controller/command" joint_trajectory)

      (setq i (+ i 1))
      (ros::sleep)
    ))
  (exit))

(send_joint_position)

```

ishiguroJSK, 2020-11-05:
```
[ INFO] [1604587295.085277145]: send_joint_position function !
[ INFO] [1604587295.094741485]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587296.276442841, 1098.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587296.276965485, 1098.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587297.274563135, 1099.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587298.274273986, 1100.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587299.274260138, 1101.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587300.279208179, 1102.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587301.284935697, 1103.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]
[ INFO] [1604587302.288465178, 1104.340000000]: joint [((/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2))]

```
１Hzで動きはしているみたいですが，
期待している動作をさせるためには，

`(send point :positions (list (/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2)))`
とか
`(send joint_trajectory :points (list point))`
とかに直してみたらどうでしょう？

atnarita, 2020-11-05:
ありがとうございます。
```(send point :positions (list (/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2)))```に直しました。
自分のパソコンで動かすと写真のように止まってしまうのですが何が原因なのでしょうか。
```rostopic echo [/fullbody_controller/coand```で見ても反応がありません。

[image]

atnarita, 2020-11-05:
```rosrun```を使ったら上手くできました。

closed 2020-11-05 by atnarita