jsk-enshu/robot-programming #372: フィボナッチについて

ummaaa, 2020-11-06:
11/2資料10ページのfibonacci-client.lはどこにおけばいいのでしょうか。色んな所を試したのですが、
rosrun beginner_tutorials fibonacci-client.l
とすると以下のようなエラーが出てしまいます。roscoreとsetup.bashはやりました。

[rosrun] Couldn't find executable named fibonacci-client.l below /home/mech-user/catkin_ws/src/beginner_tutorials
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/mech-user/catkin_ws/src/beginner_tutorials/euslisp/scripts/fibonacci-client.l
[rosrun]   /home/mech-user/catkin_ws/src/beginner_tutorials/euslisp/fibonacci-client.l
[rosrun]   /home/mech-user/catkin_ws/src/beginner_tutorials/fibonacci-client.l

k-okada, 2020-11-06:
chmod u+x して実行ビットを付与してみてください
--
◉ Kei Okada

ummaaa, 2020-11-06:
ありがとうございます。

closed 2020-11-06 by ummaaa