jsk-enshu/robot-programming #373: 11/02課題１の，/camshift/track_box のサブスクライブの仕方

ketaro-m, 2020-11-06:
11/02の課題1で，10/12の課題3で作った「CamShift Demo を用いて色領域を認識し，その方向を向く演習ロボットの行動プログラム」"[face_recognizedColor.py](https://gist.github.com/ketaro-m/918c08894117118adf7b79b24daccb07)"をeuslispに翻訳する形で"[face_recognizedColor.l](https://gist.github.com/ketaro-m/918c08894117118adf7b79b24daccb07)"を作成しようとしたのですが，以下のようなエラーが出てしまいます．
"/camshift/track_box"というトピックをコールバック関数内でサブスクライブしてスロット変数"rect"としてセットし，ループ内でそのメンバ変数"rect"にアクセスしようとしているのですが，その変数が存在しないというエラーが出てしまいます．ソースコードにあるように`ros::ros-info`で変数を見ようとするとコマンド出力の一番上のようにnilと出てしまうので，おそらくコールバック関数が呼ばれてスロット変数rectにセットされる前にループでアクセスしようとしてエラーが出てしまうのだと思います．pythonで書いたときはこのようなエラーは出なかったのですが，何か解決策はあるのでしょうか．
また，トピックのサブスクライブ周り(特にトピックのメンバ変数へのアクセスの仕方)に自信がないのですが，ソースコードのような書き方であっているのでしょうか．

[ソースコードとエラー出力]((https://gist.github.com/ketaro-m/918c08894117118adf7b79b24daccb07))

[image]

k-okada, 2020-11-08:
おそくなりました．
```
$ rosmsg show opencv_apps/RotatedRectStamped
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
opencv_apps/RotatedRect rect
  float64 angle
  opencv_apps/Point2D center
    float64 x
    float64 y
  opencv_apps/Size size
    float64 width
    float64 height
```
なので，この型の変数をmsgとすると`msg->rect->center->x` というデータ構造になります．
なので，pythonだと，
```
>>> from opencv_apps.msg import RotatedRectStamped                                            
>>> msg = RotatedRectStamped()
>>> msg.rect.center.x
0.0
```
roseusだと
```
$ (setq msg (instance opencv_apps::RotatedRectStamped :init)
#<opencv_apps::rotatedrectstamped #X55bee490a620>
$ (send msg :rect :center :x)
0.0
$ (send (send (send msg :rect) :center) :x)
0.0
```
ですね．

ketaro-m, 2020-11-08:
お返事ありがとうございます．そのようにしたらアクセスできました．
またnilの問題も，whileの中にif文を入れることで解決できました

closed 2020-11-08 by ketaro-m