jsk-enshu/robot-programming #374: 11/2 課題１について

kohtaro246, 2020-11-07:
以下のkadai3.pyに似たプログラムを作ろうとしたのですが、1_alt.lを実行すると以下のようなエラーが出ました。どのように解決すればよいでしょうか。ちなみにwhile文の中身をすこし変えてcallback関数に入れて実行すると思い通りに動いたので、callback関数で代入した変数rectがwhile文の中で参照できていないせいなのではないかとおもっています。rectをどのようにグローバル変数化すればいいでしょうか。
コード：https://gist.github.com/kohtaro246/d5cc7cfcf92733ce0e3a921fb3b2e5b7
エラー：
[image]

k-okada, 2020-11-08:
```
;;define callback function
(setq rect nil)
(defun cb (msg)
  (setq rect (send msg :rect)))
```
みたいにして大域変数を指定できます（main関数の外に変数があれば大域変数に成るのは
C言語でもPythonでも同じ）
この場合，while文の中で参照される大域変数で`rect`が最初はnilになっている可能性があるので，
```
(if rect
  (let ((center (send rect :center)))
    (let ((x (send center :x)))
      (if (= x 0.0)
	  (send ang :z 0.3)))
```
とするか，大域変数の初期化を
```
(setq rect (instance opencv_apps::RotatedRectStamped :init))
```
とします．Pythonの方のコードはこの書き方になっています．

ただ，どちらの方法も，対象物を認識した時間があったあと，それを見失った場合でも，過去に認識した情報が大域変数に入ったままですので，見失ったことがわからず動いてしまいます．それへの対応したプログラムが下のコードになります．
https://github.com/k-okada/robot-programming/blob/add_answers/dxl_armed_turtlebot/scripts/track_box_to_cmd_vel_3.py

kohtaro246, 2020-11-08:
ご丁寧に解説いただきありがとうございます。

closed 2020-11-08 by kohtaro246