jsk-enshu/robot-programming #375: 11/2 課題3について

tkmtnt7000, 2020-11-07:
課題3について以下のコードのようにして実行したところ、以下のようにエラーが出ました。
1行ずつ試していくと(ros::publish "/fullbody_controller/command" joint_trajectory)の部分でエラーを吐いていましたが、解決法がわからないのでどの部分がおかしいか教えていただけますと嬉しいです。

プログラム
`#!/usr/bin/env roseus
(ros::roseus-add-msgs "trajectory_msgs")                          
                                                                                                                
(ros::roseus "send_joint_position")                                           
(ros::advertise "/fullbody_controller/command" trajectory_msgs::JointTrajectory 1)
(ros::sleep)                                                                  
(setq joint_trajectory (instance trajectory_msgs::JointTrajectory :init))     
(send (send joint_trajectory :header) :stamp (ros::time-now))                                                                             
(send joint_trajectory :joint_names (list 'arm_joint1 'arm_joint2 'arm_joint3 'arm_joint4 'arm_joint5 'arm_joint6))                                        

;;本当は10/12のようにfor文を回したかったがappendする方法がわからない
(setq point (instance trajectory_msgs::JointTrajectoryPoint :init))                                                                           
(send point :positions (list (/ pi 2) 0 (/ pi 4) 0 (/ pi 2) (/ pi 2)))      
(setq duration (instance ros::duration :init 1.0))                         
(send point :time_from_start duration)
(send joint_trajectory :points point) 
                    
(ros::publish "/fullbody_controller/command" joint_trajectory)
(ros::sleep)`

結果
`3.irteusgl$ load "assignment_3_1102.l"
[ WARN] [1604751423.805978658, 5997.950000000]: ROSEUS is already installed as /send_joint_position_1604751388820136093
[ WARN] [1604751423.808603299, 5997.950000000]: topic /fullbody_controller/command already advertised
Call Stack (max depth: 20):
  0: at (length ros::x)
  1: at (+ 4 (length ros::x))
  2: at (mapcar #'(lambda (ros::x) (+ 4 (length ros::x))) ros::_joint_names)
  3: at (mapcar #'(lambda (ros::x) (+ 4 (length ros::x))) ros::_joint_names)
  4: at (apply #'+ (mapcar #'(lambda (ros::x) (+ 4 (length ros::x))) ros::_joint_names))
  5: at (+ (send ros::_header :serialization-length) (apply #'+ (mapcar #'(lambda (ros::x) (+ 4 (length ros::x))) ros::_joint_names)) 4 (ap\
ply #'+ (send-all ros::_points :serialization-length)) 4)
  6: at (ros::publish "/fullbody_controller/command" joint_trajectory)
  7: at (apply #'ros::load-org-for-ros ros::fullname args)
  8: at (apply #'ros::load-org-for-ros ros::fullname args)
  9: at (let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*c\
ompile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" pac\
kage-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-mo\
dule (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (\
unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) \
(unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (retu\
rn-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (e\
rror "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
  10: at (load "assignment_3_1102.l")
  11: at #<compiled-code #X55f23eff0e18>
/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: sequence expected in (length ros::x)`

softyanija, 2020-11-08:
(send joint_trajectory :points (list point)) とすれば解決できるはずです。(list)としないとリストじゃなくて単一の変数とみなされるのかな(有識者がいたら教えてください)

k-okada, 2020-11-08:
おそくなりました．はい，損通りです．以下のようにpoints はリスト，としてメッセージで定義されています．
```
$ rosmsg show trajectory_msgs/JointTrajectory
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
string[] joint_names
trajectory_msgs/JointTrajectoryPoint[] points
  float64[] positions
  float64[] velocities
  float64[] accelerations
  float64[] effort
  duration time_from_start
```
ちなみにその部分は，
```
  (dotimes (i 5)
    (setq point (instance trajectory_msgs::JointTrajectoryPoint :init))
    ;; $ (setq point (instance trajectory_msgs::JointTrajectoryPoint :init))                  
    ;; #<trajectory_msgs::jointtrajectorypoint #X55bee493e018>                                
    ;; $ (send point :positions)                                                              
    ;; #f()                                                                                   
    ;; となるのでこれは float-vector だということがわかる．                                   
    (send point :positions (float-vector pi/2 0 (* (/ pi 4) (mod i 2)) 0 pi/2 pi/2))
    (send point :time_from_start (instance ros::duration :init (+ 1.0 i)))
    (send joint-trajectory :points
          (append (send joint-trajectory :points) (list point)))
    )
```
みたいにしてかけます．

kohtaro246, 2020-11-08:
上を参考にして書いてみたのですが、うまくいきません。解決法をご教示いただけると幸いです。
コード：https://gist.github.com/kohtaro246/6aac7629d5a66d7bec46e096ad314cda
[image]

k-okada, 2020-11-08:
```
string[] joint_names
```
なので，
```
(send joint-trajectory :joint_names (list 'arm_joint1 'arm_joint2 'arm_joint3 'arm_joint4 'arm_joint5 'arm_joint6))
```
の部分は
```
(send joint-trajectory :joint_names (list "arm_joint1" "arm_joint2" "arm_joint3" "arm_joint4" "arm_joint5" "arm_joint6"))
```
かな．

kohtaro246, 2020-11-08:
'　だとstringとして認識されないのですね。直したらうまく行きました。ありがとうございます。

tkmtnt7000, 2020-11-08:
こちらも直したら動きました。ありがとうございました。

closed 2020-11-09 by tkmtnt7000