jsk-enshu/robot-programming #376: 1102 kadai

softyanija, 2020-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

k-okada, 2020-11-08:
http://wiki.ros.org/ja/ROS/Tutorials/WritingPublisherSubscriber%28euslisp%29#CA-2c6a2af157ced92c57695e70c3f58ea8a16e93ad_33
みたいに，`(ros::spin-once)` が必要です．

これはややこしいんですが，pythonでは必要ないんですが，c++ではは必要で，roseusはC++の実装を借りているので，同じように必要になります．

https://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29#rospy_tutorials.2FTutorials.2FWritingPublisherSubscriber.CA-c82832e0d612370fe9886563f0b7f5433f6caee1_10
https://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29#roscpp_tutorials.2FTutorials.2FWritingPublisherSubscriber.CA-7875e5af8250b639212f51b1a24ec13990321f5b_103

softyanija, 2020-11-08:
(ros::sleep)の前後に書いて試してみましたが、表示は変わりませんでした。また、usrbinなどどわざとタイポしても同じエラーになってしまいました。

ishiguroJSK, 2020-11-08:
これそもそもdefunの（）が対応して無くないかな？

softyanija, 2020-11-08:
~~~
#!/usr/bin/env roseus

(ros::roseus-add-msgs "trajectory_msgs")
(ros::roseus-add-msgs "control_msgs")
(ros::roseus "send_angle")

(defun send_joint_pos_actionlib ()
  (let (goal)
    (if (not (boundp '*c*))
	(setq *c* (instance ros::simple-action-client :init
			    "/fullbody_controller/follow_joint_trajectory" control_msgs::FollowJointTrajectoryAction)))
    (send *c* :wait-for-server)
    (setq goal (instance control_msgs::FollowJointTrajectoryGoal :init))
    (setq goal_t (instance trajectory_msgs::JointTrajectory :init))
    (send goal_t :header :stamp (ros::time-now))
    (send goal_t :joint_names  (list "arm_joint1" "arm_joint2" "arm_joint3" "arm_joint4" "arm_joint5" "arm_joint6"))
    (setq i 0)
    (ros::rate 1)
    (while (ros::ok)
      (setq point (instance trajectory_msgs::JointTrajectoryPoint :init))
      (send point :positions (list (/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2)))
      (send point :time_from_start (instance ros::duration :init (+ 1.0 i)))
      (send goal_t :points
	    (append (send goal_t :points) (list point)))
      (send *c* :send-goal goal)
      (setq i (+ i 1))
      (ros::ros-info "wait for goal...")
      (send *c* :wait-for-result)
      (ros::ros-info "done")
      (ros::sleep)
      (ros::spin-once)
      )
    (exit)))

(send_joint_pos_actionlib)
~~~
上のようにしたら動くようにはなりました。ただ、1ループの時間を1sにできません。(ros::rate 1)でできるとおもっていたのですが、どうすればいいでしょうか。

k-okada, 2020-11-08:
@ishiguroJSK 最小限で再現できるコードが何か教えてほしいんだけど，
https://gist.github.com/k-okada/1c08684be1ac5077818687201fc24186
でエラーにならないので問題はない気がするけど，そうでもないのかな．

softyanija, 2020-11-08:
最初の方針のプログラムは問題なく動くようになりました、ありがとうございます！
ishiguroさんのおっしゃっていた方針の方はgoalは送れているみたいですが何故か動かなかったので引き続き頑張ってみます。

ishiguroJSK, 2020-11-08:
@k-okada
`roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch`
のあとに
`roseus test.l`
↓
```
#!/usr/bin/env roseus
(ros::roseus-add-msgs "trajectory_msgs")
(ros::roseus-add-msgs "control_msgs")
(ros::roseus "test")

(setq *c* (instance ros::simple-action-client :init "/fullbody_controller/follow_joint_trajectory" control_msgs::FollowJointTrajectoryAction))
(send *c* :wait-for-server)
(setq goal (instance control_msgs::FollowJointTrajectoryGoal :init))
(ros::rate 0.2)
(while (ros::ok)
  (send *c* :send-goal goal)
  (ros::ros-info "wait for goal...")
  (send *c* :wait-for-result)
  (ros::ros-info "done")
  (ros::sleep)
  )
```
これで正しいgoalが送れてないのは承知なのですが，だとしてもros::sleepが効かないのはおかしい気がします．
(send *c* :wait-for-result)をコメントアウトすると正しく５秒に１回プリントされます．

k-okada, 2020-11-08:
`:wait-for-result` するとそこでresultがくるまでブロックするんじゃない？

ishiguroJSK, 2020-11-08:
```
$ roseus test.l 
configuring by "/opt/ros/melodic/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; extnum ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph ___time ___pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtstl irtwrl irtpointcloud eusbullet bullet irtcollision irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x55575306f690[16374] --> 0x5557534f8c00[32748] top=3c74
irtgl irtglc irtviewer 
EusLisp 9.27( 1.2.2) for Linux64 created on ip-172-30-1-213(Wed Sep 30 02:03:40 UTC 2020)
roseus ;; loading roseus("1.7.4") on euslisp((9.27 ip-172-30-1-213 Wed Sep 30 02:03:40 UTC 2020  1.2.2))
eustf roseus_c_util [ INFO] [1604841822.797784207, 6.210000000]: wait for goal...
[ INFO] [1604841822.799365656, 6.210000000]: done
[ INFO] [1604841822.804061830, 6.220000000]: wait for goal...
[ INFO] [1604841822.814340242, 6.230000000]: done
[ INFO] [1604841822.824318099, 6.240000000]: wait for goal...
[ INFO] [1604841822.834592043, 6.250000000]: done
[ INFO] [1604841822.843393394, 6.260000000]: wait for goal...
[ INFO] [1604841822.853735665, 6.270000000]: done
[ INFO] [1604841822.863660889, 6.280000000]: wait for goal...
[ INFO] [1604841822.874113176, 6.290000000]: done
[ INFO] [1604841822.883966702, 6.300000000]: wait for goal...
[ INFO] [1604841822.900185498, 6.310000000]: done
[ INFO] [1604841822.903768262, 6.320000000]: wait for goal...
[ INFO] [1604841822.914084656, 6.330000000]: done
[ INFO] [1604841822.923958858, 6.340000000]: wait for goal...
[ INFO] [1604841822.933871712, 6.350000000]: done
[ INFO] [1604841822.943683152, 6.360000000]: wait for goal...
[ INFO] [1604841822.956990913, 6.370000000]: done
[ INFO] [1604841822.963921969, 6.380000000]: wait for goal...
[ INFO] [1604841822.973968266, 6.390000000]: done
[ INFO] [1604841822.983770751, 6.400000000]: wait for goal...
[ INFO] [1604841822.998793681, 6.410000000]: done
[ INFO] [1604841823.006537713, 6.420000000]: wait for goal...
[ INFO] [1604841823.016422148, 6.430000000]: done
[ INFO] [1604841823.026449510, 6.440000000]: wait for goal...
[ INFO] [1604841823.037488515, 6.450000000]: done
[ INFO] [1604841823.046202792, 6.460000000]: wait for goal...
[ INFO] [1604841823.056278932, 6.470000000]: done
[ INFO] [1604841823.065905310, 6.480000000]: wait for goal...
[ INFO] [1604841823.076853658, 6.490000000]: done
[ INFO] [1604841823.085681046, 6.500000000]: wait for goal...
[ INFO] [1604841823.095803689, 6.510000000]: done
[ INFO] [1604841823.113191324, 6.520000000]: wait for goal...
[ INFO] [1604841823.115755641, 6.530000000]: done
[ INFO] [1604841823.125528410, 6.540000000]: wait for goal...
[ INFO] [1604841823.138612555, 6.550000000]: done
[ INFO] [1604841823.146312213, 6.560000000]: wait for goal...
[ INFO] [1604841823.158818856, 6.570000000]: done
[ INFO] [1604841823.165695149, 6.580000000]: wait for goal...
[ INFO] [1604841823.176888933, 6.590000000]: done
[ INFO] [1604841823.185728721, 6.600000000]: wait for goal...
```
ブロックせずに一瞬で返ってきてる感あります

k-okada, 2020-11-08:
https://gist.github.com/k-okada/1c08684be1ac5077818687201fc24186 にテストを追加して，さらに，.test ファイルを作ってくれて，本当はそれをPRで送ってくれると，確認しやすいのでお願いします．

k-okada, 2020-11-08:
https://gist.github.com/k-okada/59d47a9fdebf10ce321a57d6ec2bcf2e
で `rostest fuga.test` とすると，問題ないように見えるんだけど．．．

k-okada, 2020-11-08:
なるほど，