jsk-enshu/robot-programming #379: 1109 swigでのエラーついて

softyanija, 2020-11-13:
資料10p後半のようにしてswigを使おうとすると次のようなエラーを吐かれます。何が原因なのでしょうか。
```
mech-user@ki00087:~/Documents/mech3A/soft3/1109$ swig -c++ -java ffi-class.i
mech-user@ki00087:~/Documents/mech3A/soft3/1109$ g++ -shared -fPIC -I./ -I$JAVA_HOME/include -o libffi-class_wrap.so ffi-class_wrap.cxx ffi-class.cpp -lm -lstdc++
In file included from ffi-class_wrap.cxx:178:0:
/usr/lib/jvm/java-11-openjdk-amd64/include/jni.h:45:10: fatal error: jni_md.h: そのようなファイルやディレクトリはありません
 #include "jni_md.h"
          ^~~~~~~~~~
compilation terminated.
```

ishiguroJSK, 2020-11-13:
https://stackoverflow.com/questions/37023234/fatal-error-jni-md-h-no-such-file-or-directory-include-jni-md-h/37029528

@k-okada 明示的にマシン依存のパスを通してやる必要がありそうです．

`g++ -shared -fPIC -I./ -I$JAVA_HOME/include -I$JAVA_HOME/include/linux -o libffi-class_wrap.so ffi-class_wrap.cxx ffi-class.cpp -lm -lstdc++`

softyanija, 2020-11-13:
ちゃんと動きました、ありがとうございます！