jsk-enshu/robot-programming #380: 11/9の課題に関して

kohtaro246, 2020-11-15:
Ctypesを用いて、ホスト言語をCUDA C ゲスト言語をPythonするプログラムを書いたのですが、これを課題として提出して構わないでしょうか。（実行は学科PCではできず、NVIDIAのGPUが必要になります。もしかしたらGoogle colaboratoryなどを使えば実行できるかもしれません）

ホストのコンパイル方法：
/usr/local/cuda/bin/nvcc -Xcompiler -fPIC -shared -o cuda_mul.so cuda_mul.cu

ホスト、ゲストのソースコード：
https://gist.github.com/kohtaro246/57fcc74a1076bd0681367d7d1b77af03

実行結果：
[image]

k-okada, 2020-11-15:
将来こういうのをやるときに既存のライブラリで使いたい関数が提供されていなくても，自前でCUDAを良い出せるようになってほしい，という狙いがあったので良い例だとおもいます．pythonがホストかな．
--
◉ Kei Okada

kohtaro246, 2020-11-15:
お返事ありがとうございます。
ホストとゲストを混同していました。

closed 2020-11-15 by kohtaro246