jsk-enshu/robot-programming #382: 11/9課題　SWIGのエラーについて(Eigen::MatrixXdとNumpyのndarrayの変換規則が必要という認識が正しいかどうか)

RyotaUshio, 2020-11-15:
11/9(FFI)分の課題で、SWIGを用いてC++で書いた数理最適化のプログラムをPythonから呼び出せるようにしようとしています。
C++のプログラムはこちらになります。
-[continuous.hpp](https://github.com/RyotaUshio/optimization/blob/main/tmp/continuous.hpp)
-[continuous.cpp](https://github.com/RyotaUshio/optimization/blob/main/tmp/continuous.cpp)
Python側ではこんな感じで動かしたいと思っています。これはf(x)=x^2+2x+1の停留点を最急降下法で求めるサンプルプログラムです。
-[test.py](https://github.com/RyotaUshio/optimization/blob/main/swig/test.py)

これを実現するために、次のようなインターフェイスファイルを書きました。
-[continuous.i](https://github.com/RyotaUshio/optimization/blob/main/swig/continuous.i)

[Makefile](https://github.com/RyotaUshio/optimization/blob/main/swig/Makefile)を用意して`make`すると`_continuous.so`が生成され、これをPythonでインポートするところまではできました。また、コンストラクタが特に引数を必要としないような自作クラスのオブジェクトも生成できました。

ところが、さきほどの`test.py`を実行すると以下のエラーが出ます。
```
$ python test.py
Traceback (most recent call last):
  File "test.py", line 27, in <module>
    f = cont.objFunc(func, grad, hesse)
  File "/home/ryotaushio/optimization/swig/continuous.py", line 115, in __init__
    this = _continuous.new_objFunc(*args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_objFunc'.
  Possible C/C++ prototypes are:
    Continuous::objFunc::objFunc(funcType)
    Continuous::objFunc::objFunc(funcType,gradType)
    Continuous::objFunc::objFunc(funcType,gradType,hesseType)
```
このエラーは`test.py`の27行目で`objFunc`という最適化の目的関数を表すクラスのオブジェクトを生成するときに発生しているのですが、エラーを見る限りでは、引数の型がおかしいというようなことを言われているのだと思います。ここで`funcType`・`gradType`・`hesseType`というのは`continuous.hpp`の11-13行目で
```
using funcType = std::function<double(Eigen::VectorXd)>;
using gradType = std::function<Eigen::VectorXd(Eigen::VectorXd)>;
using hesseType = std::function<Eigen::MatrixXd(Eigen::VectorXd)>;
```
と宣言されているエイリアスで、いずれもEigenの`VectorXd`を引数にとり、`funcType`は`double`を、`gradType`は`VectorXd`(勾配ベクトルのつもり)を、`hesseType`は`MatrixXd`(ヘッセ行列のつもり)を返す関数オブジェクトです。

`test.py`の当該箇所の`f = cont.objFunc(func, grad, hesse)`でコンストラクタに渡しているのは、いずれもPythonのarray-likeなオブジェクトを引数にとる関数で、`func`は`float`や`np.float64`などのスカラー型を、`grad`はNumpyの1次元の`ndarray`を、`hesse`は2次元の`ndarray`を返す関数です。

この状況から推察するに、おそらくEigenのベクトルや行列とNumpy配列との変換規則のようなものを定義してやらないと、`objFunc`オブジェクトを生成することはできないのかなと思いましたが、この認識であっているでしょうか？
([こんなもの](https://github.com/rdeits/swig-eigen-numpy)も見つけました)

k-okada, 2020-11-16:
おそくなりました，これ，なかなか難しいケースだね．
Eigen自体もそうだし，std::function も対応が必要そうです．

https://stackoverflow.com/questions/24375198/error-wrapping-eigen-c-with-python-using-swig
https://stackoverflow.com/questions/32644268/how-to-use-swig-to-wrap-stdfunction-objects

k-okada, 2020-11-16:
で，この２つを頑張っても，上手く対応できるか？というところまでは自信はないです．

RyotaUshio, 2020-11-17:
ありがとうございます。相当いろいろなことを調べないと難しそうですね...。
少なくともEigenのほうはswigでなくてBoost.Python Numpy extentionというのを使うと比較的簡単にできるかもしれないので、Boost.Pythonの勉強も視野に入れて頑張ってみます。

k-okada, 2020-11-17:
そうですね．
あとは，そこまでややこしくなければ，Cに渡すときにシンプルなfloat
の配列にして渡して，Cで受け取ってC++のEigenに入れる，みたいな作戦もあります．
https://github.com/euslisp/jskeus/pull/555/files
などは，そのさくせんでやっています．
Swigなんかも，一番下ではこうなっているはずです．
--
◉ Kei Okada