jsk-enshu/robot-programming #385: wstool update について

ketaro-m, 2020-11-17:
冬学期の知能ロボット行動プログラミング演習の資料に，このワークスペースの最新バージョンを取得するのに`wstool update`を用いるように記載してあるのですが，具体的にgit上ではどのような操作がなされているのでしょうか．
資料にも

> 実態はgit pull

と記載してあるのですが，具体的にはどのブランチがpullされるのでしょうか．ローカルで`robot-system`というブランチを切ってて作業しているのですが，`gitk`でコミットツリーを見てみてもリモートブランチがマージされた形跡がないので，どのような中身になっているのか気になりました．

ishiguroJSK, 2020-11-17:
すいません．wstoolは.rosinstall.melodicに書かれているレポジトリを一括アップデート(git pull)するもので，
実はrobot-programmingのレポジトリはwstoolの管理下にありません
(robot-programmingは最初に手動でgit cloneしてもらっている)．
演習の時は「wstool updateして最新にしてください」といいましたが，それではrobot-programmingは最新にならないので，結局robot-programming以下でgit pull origin masterするか，catkinワークスペースを作り直してもらうことになりました．

ketaro-m, 2020-11-17:
そういうことなんですね，お返事ありがとうございます．
ちなみにwstool管理下にあるリポジトリの場合は，wstool updateをすると.rosinstall.melodicに書いてあるリポジトリのデフォルトブランチをローカルのHEADの位置にpullすることであり，その場合はコミットツリー上でもその形跡が残るという認識であっていますか？

ishiguroJSK, 2020-11-18:
あんまり使い倒してないから，正確な答えが出来ないんだけども，
.rosinstallにコミットハッシュが指定してあったら，そのコミットをcheckout，
ブランチ名が指定してあったらそのブランチの最新をcheckoutしているような挙動をしている．

> ローカルのHEADの位置にpullすることであり，その場合はコミットツリー上でもその形跡が残るという認識であっていますか？

ローカルの作業ブランチにリモートの指定ブランチがpullされてマージされたり，
そのマージコミットが残るようなこともなかったはず．

同じブランチの最新コミットを取得するという意味で「git pull」と表現したけども，
「リモートの指定ブランチをcheckout」とかのほうが良かったかな・・・

ketaro-m, 2020-11-18:
なるほど，そういった仕組みなんですね．
ご丁寧にお返事いただきありがとうございました．

closed 2020-11-18 by ketaro-m