jsk-enshu/robot-programming #388: 知能ロボットプログラミング、最終課題に関して

kohtaro246, 2020-11-21:
夜分遅くにすみません。
最終課題として以下のようなプログラムを実装しました。

内容：
・Turtlebotロボットが顔を探し、見つけたら顔に近づき、一定の距離に近づいたらアームを伸ばす。
・顔とロボットの距離は電子回路実習で用いた超音波センサで図るものとする。
・ロボットが回転するとき、電子回路実習で用いたステッピングモーターがロボットの回転角と同じだけ回転する。

その結果、写真のように、rostopic echo /rangeで流れてくる超音波センサのデータと、プログラムでサブスクライブすることによって得た超音波センサのデータにときに乖離が生じるようになりました。（一致するときもあります。）これを解決する方法はありますか。
（プログラムを実行すると、パソコンの動作がかなりカクカクになるので、プログラムの方でラグが起きてしまっているからなのかと推測していますが、自信がありません）

実行ファイル：https://gist.github.com/kohtaro246/00a760278e7ebf58766afe7593d7eeeb

実行方法：
以下を実行する
・roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
・roslaunch mechatrobot mechatrobot_driver.launch
・roslaunch mechatrobot sample_face_detect.launch
・rosrun mechatrobot final.py

実行画面：
[image]

ishiguroJSK, 2020-11-21:
ちょっとパッと原因は分からないけど，Rangeのheaderも併せてprintしたら，
どっちが何秒遅れてるか分かるけど，どうだろう？

あとステッピングモーターが動いているときはRangeの更新が止まる問題もあったりする

kohtaro246, 2020-11-21:
お返事ありがとうございます。やってみたところ２０秒ほど遅れているようです。
[image]

kohtaro246, 2020-11-21:
距離をprintするときはステッピングモーターを動かさない実装にしているのでステッピングモーターの影響は大丈夫だとおもいます。普通２０秒も遅れるものなのでしょうか。

ishiguroJSK, 2020-11-21:
> 普通２０秒も遅れるものなのでしょうか。

遅れないはずなのですが，演習で使ったArduinoのステッピングモーター用のライブラリが，
動作中は他の処理がブロックされる説があって，ステッピングモータが目標に到達するまでArduinoとの送受信が止まっているんじゃないかと思っています．

本当にmotor1/commandを全く送らず，ステッピングモータを全く動かさない実装にして試したかな？

kohtaro246, 2020-11-21:
プログラム66行目のpub_motor.publish(motor_command_msg)をコメントアウトすることによって全くパブリッシュされない実装にしてみても、１５秒から２０秒のラグが出てしまいます。
[image]

kohtaro246, 2020-11-21:
send_joint_pos()関数内のrospy.sleepを取り除くとうまく行きました。
夜遅くに申し訳ありませんでした。

closed 2020-11-21 by kohtaro246