jsk-enshu/robot-programming #401: 1018の発展課題rat classについて

RailgunJu, 2021-10-31:
sample programのrat.lのなかでいくつか理解できないところがあって、基礎的なものですが、
一つ目は、三行目の親クラスのところにpropertied-objectがあるんですが、なにを表してるんでしょうか。

２つ目は、
`(:prin1 (strm) (format strm "~A/~A" numer denom))`
の意味がわかりません。print文の仕組みなのでしょうか、なんでsendを使わずに自動的実行されるんですか。strmは何でしょうか。

また、setfとsetqは何が違うんですか。

k-okada, 2021-10-31:
`propertied-object` は，デフォルトの親クラスみたいなもので，Pythonだと`object`に相当するもの，という理解で良いとおもいます．
https://stackoverflow.com/questions/4015417/why-do-python-classes-inherit-object

EusLispは関数でもメンバ関数でも最後に評価（実行）された結果が，返り値になります．したがって，`(format ...)` の返り値がこのメンバ関数の返り値になります．
`format` はpythonの`format` や Cの`printf` と同じようにフォーマットする機能で，以下のように使います．
```
(format t "~A/~7,2f/~f~%" 10 1.234 1.234)
10/   1.23/1.2
```
第一引数の`t`は C言語の`fprintf(s, "hello world\n")`の第一引数みたいなもので，フォーマットの出力を示します．`t`だと標準出力です．それ以外はファイルなどが有りえます．面白い使い方としては，
```
(setq a (format nil "~A/~7,2f/~f~%" 10 1.234 1.234))
(print a)
```
などとできます．

`setf` は変数の一部を変更することができます．以下のようにリストやベクトルの要素を直接変更する時に利用します．
```
$ (setq a (list 1 2 3))
(1 2 3)
$ (print (elt a 1))
2
$ (setf (elt a 1) 10)
10
115.irteusgl$ (print a)
(1 10 3)
$ (setq b #f(1 2 3))
#f(1.0 2.0 3.0)
$ (print (elt b 1))
2.0
$ (setf (elt b 1) 10)
10
$ (print b)
#f(1.0 10.0 3.0)
```
===

http://euslisp.github.io/EusLisp/jmanual-node4.html
```
(defclass object :super NIL :slots ())
(defclass propertied-object :super object
     :slots (plist)) ;property list
````
http://euslisp.github.io/EusLisp/jmanual-node7.html
```
property-listを持つオブジェクトを定義する。 他のCommon Lispと違って、 EusLispは、たとえ、symbolでなかったとしても、property-listを持つ propertied-objectを継承するどんなオブジェクトも 許可する。
```

RailgunJu, 2021-10-31:
理解しました。
ご対応ありがとうございました。