jsk-enshu/robot-programming #402: 第2回の発展課題についての質問です

Masanori-Konishi, 2021-10-31:
第2回の発展課題で、カメラに向かって顔を動かした方向にロボットが回転するようなプログラムを書いて、一応成功したのですが、顔が画面から外れたりすると以下のようなエラーがいくつも出ました。原因と解決策がよくわからないので教えていただきたいです。今更で申し訳ないのですが、よろしくお願いいたします。

エラー：
[ERROR] [1635698725.281681, 13776.110000]: bad callback: <bound method track_face_to_cmd_vel.cb of <__main__.track_face_to_cmd_vel instance at 0x7fd8ea168fa0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/mech-user/enshu_ws/src/konishi/scripts/kadai2-3.py", line 24, in cb
    if area > 10 * 10: # 認識結果面積が一定値以上のときは res に登録
UnboundLocalError: local variable 'area' referenced before assignment

ソースコード：
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import rospy
from opencv_apps.msg import RotatedRectStamped
from opencv_apps.msg import FaceArrayStamped
from geometry_msgs.msg import Twist
class track_face_to_cmd_vel:
	res = None ## メンバ変数として定義
	pub = None
	def __init__(self):
		self.res = FaceArrayStamped()
		rospy.init_node('client')
		## コールバック関数を self.cb として，メンバ関数を登録
		##顔認識参考　http://docs.ros.org/en/kinetic/api/opencv_apps/html/msg/Rect.html
		rospy.Subscriber('/face_detection/faces', FaceArrayStamped, self.cb)
		self.pub = rospy.Publisher('/cmd_vel', Twist)
	def cb(self, msg):
		## 画像処理の結果を取得
		if len(msg.faces)>0:
			area = msg.faces[0].face.width * msg.faces[0].face.height
			rospy.loginfo("area = {}, center = ({}, {})".format(
				area, msg.faces[0].face.x, msg.faces[0].face.y))
		
		if area > 10 * 10: # 認識結果面積が一定値以上のときは res に登録
			self.res = msg

	def loop(self):
		rate = rospy.Rate(10)
		while not rospy.is_shutdown():
			cmd_vel = Twist()
			if len(self.res.faces)>0:
				cmd_vel.angular.z = 0.5*(self.res.faces[0].face.x-320)/320
			rospy.loginfo("\t\t\t\t\t\tpublish {}".format(cmd_vel.angular.z))
			self.pub.publish(cmd_vel)
			rate.sleep()

if __name__ == '__main__': # メイン文．
	try:
		obj = track_face_to_cmd_vel() # track_box_to_cmd_vel オブジェクトを生成
		obj.loop() # obj.loop() メンバ関数内で無限ループとなる．
	except rospy.ROSInterruptException: pass # エラーハンドリング

k-okada, 2021-11-01:
```
	def cb(self, msg):
		## 画像処理の結果を取得
		if len(msg.faces)>0:
			area = msg.faces[0].face.width * msg.faces[0].face.height
			rospy.loginfo("area = {}, center = ({}, {})".format(
				area, msg.faces[0].face.x, msg.faces[0].face.y))
		
		if area > 10 * 10: # 認識結果面積が一定値以上のときは res に登録
			self.res = msg
```
`UnboundLocalError: local variable 'area' referenced before assignment` で，`area`という変数の値が決まっていないのに使われているというエラーです．
見てみると，`area`は`if len(msg.faces)>0:` の条件のときに値が決まっていますが，そうでない場合の値は決まっていないので，次の行の`if area > 10 * 10:` でエラーになります．
これを回避するには，

```
	def cb(self, msg):
		## 画像処理の結果を取得
		if len(msg.faces)>0:
			area = msg.faces[0].face.width * msg.faces[0].face.height
			rospy.loginfo("area = {}, center = ({}, {})".format(
				area, msg.faces[0].face.x, msg.faces[0].face.y))
		else:
                        area = 0
		if area > 10 * 10: # 認識結果面積が一定値以上のときは res に登録
			self.res = msg
```
などと，`if len(msg.faces)>0:` の条件の真偽にかかわらず，`area`の値を決めてあげるか，
```
	def cb(self, msg):
		## 画像処理の結果を取得
		if len(msg.faces)>0:
			area = msg.faces[0].face.width * msg.faces[0].face.height
			rospy.loginfo("area = {}, center = ({}, {})".format(
				area, msg.faces[0].face.x, msg.faces[0].face.y))
		        if area > 10 * 10: # 認識結果面積が一定値以上のときは res に登録
			        self.res = msg
```
と，`area`の値が決まる，`if len(msg.faces)>0:` の条件が真のときだけ，` if area > 10 * 10:`の文に入るようにします．

Masanori-Konishi, 2021-11-14:
ありがとうございます！