jsk-enshu/robot-programming #403: 1101の対象物の位置を取得するプログラムについて

AoN-11, 2021-11-05:
資料の6.1から実行していくと`(send *l* :lookup-transform "/camera_rgb_optical_frame" "/test_object" (ros::time 0))`で`(send *l* :lookup-transform "/camera_rgb_optical_frame" "/test_object" (ros::time 0))`というエラーになります。frames.pdf でも"/test_object"の座標が無いことを確認しました。
関係があるのかわかりませんが、配布されたexample.launchを立ち上げるとjointの部分に`No p gain specified for pid.`というエラーも出ます。

自分で変更した部分は特に無いと思うので、どのファイルをどう修正すればよいか教えていただきたいです。

```

$ roslaunch example.launch
... logging to /home/mech-user/.ros/log/e5a847bc-3dc9-11ec-919d-20791855a5cd/roslaunch-ki00115-21410.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Deprecated: xacro tag 'kobuki' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/mech-user/enshu_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

xacro.py is deprecated; please use xacro instead
started roslaunch server http://ki00115:35707/

SUMMARY
========

PARAMETERS
 * /camera/rgb/display: 1
 * /camera/rgb/frame_id: camera_rgb_optica...
 * /camera/rgb/grid0_size_x: 7
 * /camera/rgb/grid0_size_y: 4
 * /camera/rgb/objectdetection_tf_publisher/use_simple_tf: True
 * /camera/rgb/rect0_size_x: 0.02
 * /camera/rgb/rect0_size_y: 0.02
 * /camera/rgb/rotation0: 1.0 0.0 0.0 0.0 -...
 * /camera/rgb/single: 1
 * /camera/rgb/translation0: 0 0 0
 * /camera/rgb/type0: test_object
 * /cmd_vel_mux/yaml_cfg_file: /home/mech-user/e...
 * /depthimage_to_laserscan/output_frame_id: camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /fullbody_controller/action_monitor_rate: 100
 * /fullbody_controller/constraints/arm_joint1/goal: 0.02
 * /fullbody_controller/constraints/arm_joint2/goal: 0.02
 * /fullbody_controller/constraints/arm_joint3/goal: 0.02
 * /fullbody_controller/constraints/arm_joint4/goal: 0.02
 * /fullbody_controller/constraints/arm_joint5/goal: 0.02
 * /fullbody_controller/constraints/arm_joint6/goal: 0.02
 * /fullbody_controller/constraints/goal_time: 0.5
 * /fullbody_controller/constraints/stopped_velocity_tolerance: 0.02
 * /fullbody_controller/gains/arm_joint1/d: 0.0
 * /fullbody_controller/gains/arm_joint1/i: 0.0
 * /fullbody_controller/gains/arm_joint1/p: 25.0
 * /fullbody_controller/gains/arm_joint2/d: 0.0
 * /fullbody_controller/gains/arm_joint2/i: 0.0
 * /fullbody_controller/gains/arm_joint2/p: 25.0
 * /fullbody_controller/gains/arm_joint3/d: 0.0
 * /fullbody_controller/gains/arm_joint3/i: 0.0
 * /fullbody_controller/gains/arm_joint3/p: 25.0
 * /fullbody_controller/gains/arm_joint4/d: 0.0
 * /fullbody_controller/gains/arm_joint4/i: 0.0
 * /fullbody_controller/gains/arm_joint4/p: 25.0
 * /fullbody_controller/gains/arm_joint5/d: 0.0
 * /fullbody_controller/gains/arm_joint5/i: 0.0
 * /fullbody_controller/gains/arm_joint5/p: 25.0
 * /fullbody_controller/gains/arm_joint6/d: 0.0
 * /fullbody_controller/gains/arm_joint6/i: 0.0
 * /fullbody_controller/gains/arm_joint6/p: 25.0
 * /fullbody_controller/joints: ['arm_joint1', 'a...
 * /fullbody_controller/state_publish_rate: 100
 * /fullbody_controller/stop_trajectory_duration: 0
[... 94 lines cut ...]
[ERROR] [1636075838.616518745, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint2
[ERROR] [1636075838.618521936, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint3
[ERROR] [1636075838.621905853, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint4
[ERROR] [1636075838.623469228, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint5
[ERROR] [1636075838.625025345, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint6
[ERROR] [1636075838.626974213, 0.020000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/gripper_joint
[urdf_spawner-7] process has finished cleanly
log file: /home/mech-user/.ros/log/e5a847bc-3dc9-11ec-919d-20791855a5cd/urdf_spawner-7*.log
[ INFO] [1636075838.634520976, 0.020000000]: Loaded gazebo_ros_control.
[INFO] [1636075838.645486, 0.030000]: wait_for_service(/controller_manager/load_controller): finally were able to contact [rosrpc://ki00115:37679]
[INFO] [1636075838.647520, 0.030000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1636075838.652058, 0.030000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1636075838.657301, 0.040000]: Loading controller: joint_state_controller
[INFO] [1636075838.675459, 0.060000]: Loading controller: fullbody_controller
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[INFO] [1636075839.191736, 0.560000]: Loading controller: gripper_controller
[INFO] [1636075839.282058, 0.650000]: Controller Spawner: Loaded controllers: joint_state_controller, fullbody_controller, gripper_controller
[INFO] [1636075839.291960, 0.670000]: Started controllers: joint_state_controller, fullbody_controller, gripper_controller
[tuck_arm-12] process has finished cleanly
log file: /home/mech-user/.ros/log/e5a847bc-3dc9-11ec-919d-20791855a5cd/tuck_arm-12*.log

```

k-okada, 2021-11-05:
`checkerboard_detector` はcheckerboardを認識してはじめて `test_object`がでてきますなので，以下の画像みたいにまず何らかの方法で，ロボットの視野にチェッカーボードが入るようにしてください．

人間の目にはチェッカーボードが映っていても，`checkerboard_detector`プログラムが認識しているか，というのは別問題で，下の画像みたいに緑のマークが映っていると認識できています．

jointの部分にNo p gain specified for pid.というエラーも出るのは，無視してもらって構いません．

frames.pdf でも"/test_object"の座標を確認するのは，良いデバッグで，だいぶ理解している，ということが伝わってきます．

[image]

nakane11, 2021-11-05:
解決しました。ありがとうございます。

closed 2022-02-24 by nakane11