jsk-enshu/robot-programming #404: 1025の課題で部屋の部品の位置とサイズについて

RailgunJu, 2021-11-06:
部品の大きさを知りたいとき、どう調べればいいでしょうか。
例えば、掴んだものをテーブルに置かせたいとき、テーブルの高さとか。

また、部屋内の代表的座標（cook-spotなど）以外に行かせたいところがあったら、その座標をどう調べればいいでしょうか。
`(send *room* :spots)`で出てきた
`#<cascaded-coords #X558e35a69e08 cook-spot  1100.0 1600.0 0.0 / 3.142 0.0 0.0>`の3.142 0.0 0.0は何を表しますか。

最後に、`:move-to`と`:locate`の区別はなんですか。それぞれworld座標とlocal座標で動くという理解であってますか。

k-okada, 2021-11-07:
おそくなりました．
```
irteusgl$ (send (make-coords) :rotate pi/2 :z)
#<coordinates #X55f208d6f138  0.0 0.0 0.0 / 1.571 0.0 0.0>
irteusgl$ (send (make-coords) :rotate pi/2 :y)
#<coordinates #X55f208d6c168  0.0 0.0 0.0 / 0.0 1.571 0.0>
irteusgl$ (send (make-coords) :rotate pi/2 :x)
#<coordinates #X55f208d6be38  0.0 0.0 0.0 / 0.0 0.0 1.571>
```
となります．回転行列をZYXオイラー角で表したときに，値になるかと思います．c.f https://github.com/jsk-ros-pkg/jsk_roseus/pull/662#issuecomment-916604114
＝
`move-to` は引数に座標系を使います，`locate` の引数は位置ベクトルです．
つまり，
```
irteusgl$ (setq c (make-coords))
#<coordinates #X55f208d6a8e0  0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (send c :move-to (make-coords :pos #f(0 0 100)))
#<coordinates #X55f208d6a8e0  0.0 0.0 100.0 / 0.0 0.0 0.0>
```
と，
```
irteusgl$ (setq c (make-coords))                                                                                                         
#<coordinates #X55f208d69740  0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (send c :locate #f(100 0 0))
#<coordinates #X55f208d69740  100.0 0.0 0.0 / 0.0 0.0 0.0>
```
が同じ結果になります．
world/local は第二引数で指定します．デフォルトは`:local`になっています．上のコードに続いて
irteusgl$ (send c :locate #f(100 0 0))
#<coordinates #X55f208d69740  200.0 0.0 0.0 / 0.0 0.0 0.0> ;; $(200 0 0) に移動
irteusgl$ (send c :locate #f(100 0 0) :local)
#<coordinates #X55f208d69740  300.0 0.0 0.0 / 0.0 0.0 0.0> ;; #f(300 0 0)に移動
irteusgl$ (send c :locate #f(100 0 0) :world)
#<coordinates #X55f208d69740  100.0 0.0 0.0 / 0.0 0.0 0.0> ;; #f(100 0 0)に移動
```
となります．

http://euslisp.github.io/EusLisp/jmanual-node16.html

RailgunJu, 2021-11-07:
なるほど、ありがとうございます！