jsk-enshu/robot-programming #405: 11/08　課題について

takitaharuto, 2021-11-08:
「sudo apt install ros-melodic-pddl-planner」としてpddl-plannerをインストールした後に、

「cp $(rospack find pddl_planner)/demos/temporal/coe-*.pddl」とすると
「cp: 通常ファイル '/opt/ros/melodic/share/pddl_planner/demos/temporal/coe-problem.pddl' を作成できません: 許可がありません」となり、うまく実行できませんでした。

また、「-f」オプションを指定して
「cp -f $(rospack find pddl_planner)/demos/temporal/coe-*.pddl」としても、
「cp: '/opt/ros/melodic/share/pddl_planner/demos/temporal/coe-problem.pddl' を削除できません: 許可がありません」となってしまいます。

mkogw, 2021-11-08:
私の環境では
`cp $(rospack find pddl_planner)/demos/temporal/coe-*.pddl ./`
（最後に「./」をつけ加える）
とすればカレントディレクトリに
`coe-problem.pddl`
`coe-domain.pddl`
がコピーされたので、これで解決するかと思います。

k-okada, 2021-11-08:
ごめんなさい．そうです．資料からピリオドが抜けていて@mkogw さんの指摘が正しいです．

takitaharuto, 2021-11-09:
うまく行きました！ありがとうございます。