jsk-enshu/robot-programming #408: デモプログラムのインストールが不可能

PozihameBaystar, 2021-11-09:
sudo apt install ros-melodic-navigation-stage
とすると、

この操作後に追加で 5,746 kB のディスク容量が消費されます。
エラー:1 http://packages.ros.org/ros/ubuntu bionic/main amd64 ros-melodic-navigation-stage amd64 0.2.4-1bionic.20210206.042541
  404  Not Found [IP: 2600:3404:200:237::2 80]
E: http://packages.ros.org/ros/ubuntu/pool/main/r/ros-melodic-navigation-stage/ros-melodic-navigation-stage_0.2.4-1bionic.20210206.042541_amd64.deb の取得に失敗しました  404  Not Found [IP: 2600:3404:200:237::2 80]
E: いくつかのアーカイブを取得できません。apt-get update を実行するか --fix-missing オプションを付けて試してみてください。

とエラーを吐いてしまい、navigationをインストールすることが出来ません。apt-get updateも--fix-missingも試し、ホームディレクトリや作成したワークスペース等、様々な場所でも試してみたのですがですが、同じエラーを吐いてしまいます。

また、nextage-gazebo等他の物もいくつか試してみたのですが、全て同じようなエラーが出てしまいます。

原因及び解決策としてはどのようなものがありますでしょうか？

k-okada, 2021-11-09:
sudo apt update としてから試してみてください．
--
◉ Kei Okada

PozihameBaystar, 2021-11-09:
sudo apt update を試したところ、公開鍵が不足している事が判明し、公開鍵を追加したところ無事インストール/起動出来ました。
お忙しい中ありがとうございました。