jsk-enshu/robot-programming #409: teleop_twist_keyboard.pyがshareに無い

PozihameBaystar, 2021-11-09:
rosrun teleop_twist_keyboard teleop_twist_keyboard.py

でtwist_keyboardを起動しようとしたところ、

[rosrun] Couldn't find executable named teleop_twist_keyboard.py below /opt/ros/melodic/share/teleop_twist_keyboard

とエラーが出てしまいました。ファイル検索をしたところ、teleop_twist_keyboard.pyが何故かlibフォルダ内のteleop_twist_keyboardフォルダにあり、

export ROS_PACKAGE_PATH=/opt/ros/melodic/lib

とpathを変更することで対処しようとしたのですが、これだとteleop_twist_keyboard自体が検出されず、エラーになってしまいます。

ros-melodic-teleop-twist-keyboard は最新バージョンです。

どうすれば解決出来ますでしょうか。

k-okada, 2021-11-10:
これは不思議な仕様なんだけど，`rosrun`は`/opt/ros/melodic/share/teleop_twist_keyboard` と `/opt/ros/melodic/lib/teleop_twist_keyboard`の両方を見に行きつつも，エラーメッセージでは教えてくれないのでファイルの位置は問題ありません．
あとは，ちゃんと実行ビットが立っていればよいはずで以下のように確認してみてください．
```
$ ls -al /opt/ros/melodic/lib/teleop_twist_keyboard/
total 52
drwxr-xr-x   2 root root  4096 Nov 10 12:09 .
drwxr-xr-x 156 root root 36864 Nov  8 09:45 ..
-rwxr-xr-x   1 root root  5830 Sep 22 05:31 teleop_twist_keyboard.py
```
`rosrun --debug teleop_twist_keyboard teleop_twist_keyboard.py` も試してみてください．

PozihameBaystar, 2021-11-10:
rosrun --debug teleop_twist_keyboard teleop_twist_keyboard.py

も試してみたのですが、今度は次のようなエラーが起きてしまいました。

[rosrun] Looking in catkin libexec dirs: 
/opt/ros/melodic/lib/teleop_twist_keyboard
/opt/ros/melodic/share/teleop_twist_keyboard
[rosrun] Looking in rospack dir: /opt/ros/melodic/share/teleop_twist_keyboard
[rosrun] Searching for teleop_twist_keyboard.py with permissions /111
[rosrun] Running  /opt/ros/melodic/lib/teleop_twist_keyboard/teleop_twist_keyboard.py
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/teleop_twist_keyboard/teleop_twist_keyboard.py", line 7, in <module>
    import roslib; roslib.load_manifest('teleop_twist_keyboard')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
  File "/usr/lib/python2.7/dist-packages/rospkg/__init__.py", line 42, in <module>
    from .rospack import expand_to_packages, get_package_name, \
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 37, in <module>
    from xml.etree.cElementTree import ElementTree
  File "/home/mech-user/anaconda3/lib/python3.8/xml/etree/cElementTree.py", line 3, in <module>
    from xml.etree.ElementTree import *
  File "/home/mech-user/anaconda3/lib/python3.8/xml/etree/ElementTree.py", line 96, in <module>
    import re
  File "/home/mech-user/anaconda3/lib/python3.8/re.py", line 145, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'

enumをPython3から取り除こうとしたのですが、

(base) mech-user@KI00259:~$ sudo pip3 uninstall enum34
The directory '/home/mech-user/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Cannot uninstall requirement enum34, not installed

となってしまい、取り除くことも出来ません。

pip install enum-compat

を導入してみたのですが、未だ解決に至りません。

PozihameBaystar, 2021-11-10:
ありがとうございます。--debugを付けることでteleop_twist_keyboard自体はは読み込まれるようになったようです。ただ、今度は次のようなエラーが出てしまいました。

[rosrun] Looking in catkin libexec dirs: 
/opt/ros/melodic/lib/teleop_twist_keyboard
/opt/ros/melodic/share/teleop_twist_keyboard
[rosrun] Looking in rospack dir: /opt/ros/melodic/share/teleop_twist_keyboard
[rosrun] Searching for teleop_twist_keyboard.py with permissions /111
[rosrun] Running  /opt/ros/melodic/lib/teleop_twist_keyboard/teleop_twist_keyboard.py
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/teleop_twist_keyboard/teleop_twist_keyboard.py", line 7, in <module>
    import roslib; roslib.load_manifest('teleop_twist_keyboard')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ModuleNotFoundError: No module named 'rospkg'

ROSwikiを参考にrospkgをインストールしようとしたのですが、既に最新版のがインストールされていました。ですが、依然としてrospkgが無いものとして扱われてしまいます。

どう解決すればいいでしょうか。

k-okada, 2021-11-11:
> (base) ***@***.***:~$ sudo pip3 uninstall enum34

.bashrc からアナコンダ関係の設定をコメントアウトしてください．
ROSを使うときは source ... setup.bash とするように，anacondaを使うときは，そのコメントアウトした部分をファイルにして
source ... anaconda.bash みたいにして使ってください．

PozihameBaystar, 2021-11-11:
上手く行きました。ありがとうございます。