jsk-enshu/robot-programming #412: 11/15 c++言語におけるFFI（資料ｐ３〜）　リンクができない

fryofru, 2021-11-18:
ffi-c++.cppとffi-c++-main.cというファイルを作り、それぞれ.oファイルを作ったのち、リンクするために
$ gcc -o ffi-c++-main ffi-c++.o ffi-c++-main.o　-lm
を実行すると以下のようなエラーがでます。

mech-user@ki00128:~/3A_robotsystem(pass:soft3)/7$ gcc -o ffi-c++-main ffi-c++.o ffi-c++-main.o -lm
ffi-c++.o: 関数 `hello' 内:
ffi-c++.cpp:(.text+0x3c): `std::cout' に対する定義されていない参照です
ffi-c++.cpp:(.text+0x41): `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)' に対する定義されていない参照です
ffi-c++.cpp:(.text+0x53): `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)' に対する定義されていない参照です
ffi-c++.cpp:(.text+0x5d): `std::basic_ostream<char, std::char_traits<char> >& std::endl<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&)' に対する定義されていない参照です
ffi-c++.cpp:(.text+0x68): `std::ostream::operator<<(std::ostream& (*)(std::ostream&))' に対する定義されていない参照です
ffi-c++.o: 関数 `__static_initialization_and_destruction_0(int, int)' 内:
ffi-c++.cpp:(.text+0x94): `std::ios_base::Init::Init()' に対する定義されていない参照です
ffi-c++.cpp:(.text+0xa9): `std::ios_base::Init::~Init()' に対する定義されていない参照です
collect2: error: ld returned 1 exit status

以下２つのファイルのコードです。
//ffi-c++.cpp
#include <iostream>
#include <math.h>

extern "C" {
  double sinc(double d){
    return(sin(d)/d);
  }
  void hello(char* str){
    std::cout << "hello " << str << std::endl;
  }
}

//ffi-c++-main.c

#include <stdio.h>

double sinc(double x);
void hello(char *str);

int main(){
  double f = 1.0;
  hello("world");
  fprintf(stderr,"sinc(%f) = %f\n",f,sinc(f));
}

k-okada, 2021-11-18:
ごめんなさい
授業中に説明したつもりだったんですが、資料には書いていないんですが
`gcc -o ffi-c++-main ffi-c++.o ffi-c++-main.o　-lm  -lstdc++`
としてみて下さい

fryofru, 2021-11-19:
ありがとうございます。聴き逃していました。うまく行きました。

k-okada, 2021-11-19:
良かったです
--
◉ Kei Okada