jsk-enshu/robot-programming #414: 第5回の３(p6~p9)について

Himeno59, 2021-11-22:
第５回の「３　シンプルな遠隔関数呼び出しを書く」について
./add-two-ints-server.l
と
./add-two-ints-client.l 4 5
を実行すると
Requesting 4+5
4 + 5 = 9
ではなく以下のような出力になってしまいます。どのようにしたら良いでしょうか
[image]
[image]

[image]

k-okada, 2021-11-22:
ごめんなさい，これは資料がわるくて，資料のように２つの引数の和を得るようにするには，
```
(ros::roseus-add-srvs "roseus")

(ros::roseus "add_two_ints_client2")
(ros::wait-for-service "add_two_ints")
(setq args (subseq lisp::*eustop-argument* 7))
(cond ((= (length args) 2)
       (setq x (read-from-string (elt args 0))
             y (read-from-string (elt args 1))))
      (t
       (setq x (random 10)
             y (random 20))))

(setq req (instance roseus::AddTwoIntsRequest :init))
(send req :a x)
(send req :b y)
(setq before (ros::time-now))
(setq res (ros::service-call "add_two_ints" req))
(setq after (ros::time-now))
(format t "Requesting ~d+~d~%" (send req :a) (send req :b))
(format t "~d + ~d = ~d (~A sec)~%" (send req :a) (send req :b) (send res :sum)
        (send (ros::time- after before) :to-sec))
(ros::exit)

```
としてください．
なぜ，こういう出力がでるのかは，`./add-two-ints-client.l` のコードを見てわかるぐらいになると良いなとおもいます．

Himeno59, 2021-11-22:
上手く行きました。ありがとうございます。

資料にあった~client.lは呼び出されたときにrandomでaとｂを決めていたため、上の写真のようにランダムな足し算をくりかえしていたが、~client2.lではread-from-stringで入力を受け付けてそれをaとｂにいれているというような理解で大丈夫でしょうか

k-okada, 2021-11-22:
はい．それで完璧です．
--
◉ Kei Okada

Himeno59, 2021-11-22:
理解できました。ありがとうございます。