jsk-enshu/robot-programming #415: 第

kamiharu04323, 2021-11-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kamiharu04323, 2021-11-22:
マルチエージェントプランニングの方と同様にファイルを2つに分けると動きました。失礼しました。

closed 2021-11-22 by kamiharu04323

k-okada, 2021-11-22:
よかたったです．domain ファイルにはドメイン（分野）を，problem ファイルは問題をかいて，問題が変わったらproblem ファイルだけ変えれば良いようになっているので，別のproblemを作ってみると良いとおもいます．