jsk-enshu/robot-programming #416: 第5回の6.1でlaunchした環境でrqt_guiの/cmd_velだけが反応しません

Masanori-Konishi, 2021-11-22:
第5回の6.1で、15ページの通りlaunchファイルをホームに作成して、
```
$ source ~/enshu_ws/devel/setup.bash
$ roslaunch example.launch 
```
としたところ、一見うまく行っているように見えました。その後チェッカーボードを視界に入れようとrqt_guiでロボットを移動・回転させようとしたところ、gazebo上のロボットが反応しませんでした。

一方で、ロボットアームの方はrqt_guiから操作するとgazebo上のロボットはちゃんと動いていました。また
`$ rostopic list`
したところ
```
（一部抜粋）
/cmd_vel
/cmd_vel_mux/active
/cmd_vel_mux/input/navi
/cmd_vel_mux/input/safety_controller
/cmd_vel_mux/input/switch
/cmd_vel_mux/input/teleop
/cmd_vel_mux/parameter_descriptions
/cmd_vel_mux/parameter_updates
```
のように/cmd_velは存在していて、さらに
`$ rostopic echo /cmd_vel`
とすると
```
---
linear: 
  x: -0.235
  y: 0.0
  z: 0.0
angular: 
  x: 0.0
  y: 0.0
  z: 0.0
```
のようにrqt_guiを動かすとそれに対応した値が出力されていました。以上のことから、gazeboの何かしらのエラーではないかと思ったのですが、結局原因が分からず困っています。解決策を教えていただきたいです。よろしくお願いいたします。

k-okada, 2021-11-22:
それぞれ以下のようになるか確認してください．同じだったら，`roslaunch example.launch ` の出力を全部送ってください．
```
$ rostopic info /cmd_vel
Type: geometry_msgs/Twist

Publishers: 
 * /enshu_rqt (http://p51s:43069/)

Subscribers: 
 * /cmd_vel_renamer (http://p51s:44421/)
```
```
$ rostopic find geometry_msgs/Twist
/cmd_vel_mux/input/switch
/cmd_vel_mux/input/teleop
/cmd_vel_mux/input/safety_controller
/mobile_base/commands/velocity
/cmd_vel_mux/input/navi
/cmd_vel
```

Masanori-Konishi, 2021-11-22:
`rostopic find geometry_msgs/Twist`については結果が同じだったのですが、`rostopic info /cmd_vel`については以下のように違う結果になりました

```
mech-user@ki00164:~$ rostopic info /cmd_vel
Type: geometry_msgs/Twist

Publishers: 
 * /enshu_rqt (http://ki00164:46061/)

Subscribers: None

```

Masanori-Konishi, 2021-11-22:
dxl_armed_turtlebot/launch/dxl_armed_turtlebot_gazebo.launch に削除されていた2行を追加したところ、動くようになりました。先週にturtlebotの演習があったのですが、友達によると、演習の前には動いていて課題も提出していたものの、演習後に同じように試してみたところ自分と同じ症状で動かなかったそうです。