jsk-enshu/robot-programming #422: Travis CIの設定ファイルについての質問です

nakane11, 2022-02-24:
ロボットシステムの資料を見返しており.travis.shの内容が気になったのですが,  [7行目](https://github.com/jsk-enshu/robot-programming/blob/e2622af8188d870f652315fc5a267aa273b50e4d/.travis.sh#L7)の
```
apt-get install -y sudo software-properties-common git wget sed sudo
```
でsudoを2回書く意味はあるのでしょうか?  エラーにはなりませんがインストールの指示が重複していると思います.  

Ubuntuイメージのコンテナを起動する部分から試してみたので環境は再現できていませんが,  最後のsudoを入れなくても問題ないようでした.
```
~$ sudo docker images
REPOSITORY   TAG       IMAGE ID       CREATED       SIZE
ubuntu       bionic    dcf4d4bef137   3 weeks ago   63.2MB
~$ sudo docker run -v $HOME:$HOME -e CI_SOURCE_PATH=$CI_SOURCE_PATH -it --name test ubuntu:bionic /bin/bash
root@043bb126fd34:/# cd $CI_SOURCE_PATH/
root@043bb126fd34:/home/nakane/enshu_ws/src/robot-programming# apt-get update -qq
root@043bb126fd34:/home/nakane/enshu_ws/src/robot-programming# dpkg -l|grep sudo  
root@043bb126fd34:/home/nakane/enshu_ws/src/robot-programming# apt-get install -y sudo software-properties-common git wget sed
Reading package lists...
Building dependency tree...
Reading state information...
sed is already the newest version (4.4-2).
The following additional packages will be installed:
略
Running hooks in /etc/ca-certificates/update.d...
done.
Processing triggers for dbus (1.12.2-1ubuntu1.2) ...
root@043bb126fd34:/home/nakane/enshu_ws/src/robot-programming# dpkg -l|grep sudo
ii  sudo      1.8.21p2-3ubuntu1.4    amd64        
Provide limited super user privileges to specific users
```
もう一点質問があります.    .travis.ymlの[16行目](https://github.com/jsk-enshu/robot-programming/blob/e2622af8188d870f652315fc5a267aa273b50e4d/.travis.yml#L16)`export CI_SOURCE_PATH=$(pwd)`は環境変数CI_SOURCE_PATHにサーバー上でのrobot-programmingのパスを入れる処理で,  ホストのhomeディレクトリを共有しているのでコンテナからも同じパスを使えると認識しています.  
しかし自分で同じように`-e CI_SOURCE_PATH`とすると環境変数を渡せておらず,  何が原因なのでしょうか?

**CI_SOURCE_PATH を渡せない**
```
robot-programming$ export CI_SOURCE_PATH=$(pwd)
robot-programming$ cd
~$ echo $CI_SOURCE_PATH 
/home/nakane/enshu_ws/src/robot-programming
~$ sudo docker run -v $HOME:$HOME -e CI_SOURCE_PATH -it --name test ubuntu:bionic /bin/bash
```
**CI_SOURCE_PATH を渡せる**
```
~$ sudo docker run -v $HOME:$HOME -e CI_SOURCE_PATH=$CI_SOURCE_PATH -it --name test ubuntu:bionic /bin/bash
root@e84ea49152b8:/# echo $CI_SOURCE_PATH 
/home/nakane/enshu_ws/src/robot-programming
```

**Desktop**
 - OS: Ubuntu 18.04

nakane11, 2022-02-25:
3132731 でsudoが追加されていますがよくわかりませんでした

k-okada, 2022-02-25:
@nakane11 よく見ていますね．偉いです．
１） `sudo` についてはすいません．間違いです．必要ないです．可能だったら，重複している`sudo`を消したPullRequestを送ってみてください．
２）`sudo` したときに，する前の環境変数を引き継ぐかどうか？という問題があります．https://stackoverflow.com/questions/8633461/how-to-keep-environment-variables-when-using-sudo で 500以上イイねがついているので皆，どこかで一回気になる問題ですね．
```
k-okada@p51s:~$ env | grep CI
CI_SOURCE_PATH=/home/k-okada
k-okada@p51s:~$ sudo env | grep CI
k-okada@p51s:~$ sudo -E env | grep CI
CI_SOURCE_PATH=/home/k-okada
```
みたいに`env` とすると環境変数一覧が見えて，`CI_SOURCE_PATH`が見つかるんだけど，`sudo`だけだと，その環境変数は持ち越さないので，`sudo`の中で`env`を呼んでも`CI_SOURCE_PATH`が見つかりません．これを，持ち越したい時は明示的に`sudo -E`とする必要があります．

なので，
```
k-okada@p51s:~$ sudo docker run --rm -ti -e CI_SOURCE_PATH ubuntu:bionic /bin/bash
[sudo] password for k-okada: 
root@12fe454335d9:/# echo $CI_SOURCE_PATH

root@12fe454335d9:/# exit
k-okada@p51s:~$ sudo -E docker run --rm -ti -e CI_SOURCE_PATH ubuntu:bionic /bin/bash
root@7a84febcd845:/# echo $CI_SOURCE_PATH
/home/k-okada
root@7a84febcd845:/# exit
```
みたいに`sudo -E` とすると，`sudo`上で走らせようとしている`docker run`コマンドのなかで，今見えている`CI_SOURCE_PATH`変数を`docker`に渡して，期待通り動くようになります．

`-E`なくても` -e CI_SOURCE_PATH=$CI_SOURCE_PATH `が動くのは，`sudo`を動かす前に，`$CI_SOURCE_PATH`を評価して，値に変えて，その上で，`docker run`コマンドを実行しているからです．と言うので伝わるかな．．．．
`k-okada@p51s:~$ sudo env | grep CI_SOURCE_PATH` は値が表示されないけど，`k-okada@p51s:~$ sudo echo $CI_SOURCE_PATH` は値が表示されますが，あくまでも`sudo`した先の状態で環境変数が見えているのではなくて，環境変数を値に変えてから，`sudo`が呼ばれています．

ちなみに，
```
k-okada@p51s:~$ docker run --rm -ti -e CI_SOURCE_PATH ubuntu:bionic /bin/bash
root@a78d631f299d:/# echo $CI_SOURCE_PATH
/home/k-okada
```
と`sudo`なしで，`docker`が動けば，ここまでややこしくないです．
これは，多分，自分が`docker`グループに入っていないんだとおもいます．
```
k-okada@p51s:~$ id
uid=1000(k-okada) gid=1000(k-okada) groups=1000(k-okada),4(adm),20(dialout),24(cdrom),27(sudo),30(dip),46(plugdev),113(lpadmin),129(sambashare),144(sbuild),147(libvirt),998(bumblebee),999(docker)
```
みたいにして確認してみてください．`docker`という文字列が見つからないとおもいます．対処法は，https://docs.docker.com/engine/install/linux-postinstall/ に書いてあることで，この1.はスキップできるはずで，` sudo usermod -aG docker $USER` として，念の為リブートしてみてください．多分これで自分が`docker`グループに入って`sudo`なして実行できるとおもいます．

最後に，`docker ps -a` とすると大量にコンテナが残っていないかな．これは結構なハードディスクを消費するので，あれやこれや試す時は，僕がやっているみたいに`--rm` 付けると良いです．
逆に付けないと，`docker start ..` ` docker attach ... `みたいにして何時でも環境を元に戻せます．

nakane11, 2022-02-25:
1)のPullRequestを送りました．
環境変数についても，sudoで引き継がれないのを知らなかったので説明を読んで納得できました．dockerグループへの追加も上手くいきました．

ありがとうございました

PR #423 (2022-03-02): remove duplicate installation of sudo
`apt-get install` でインストールされるパッケージのsudoが重複していたため修正しました．
Closes #422