jsk-enshu/robot-programming #427: 3dsensor.launch causes "No devices connected" error.

sktometometo, 2022-06-02:
**Describe the bug**
`turtlebot_bringup` パッケージの `3dsensor.launch` を研究室のturtlebot付属のラップトップ(hostname: KI00070) で立ち上げると以下のように "No devices connected.." というエラーが表示され、画像のトピックが出力されません。

```bash
~ $ roslaunch turtlebot_bringup 3dsensor.launch
... logging to /home/sktometometo/.ros/log/76cbde5e-e22d-11ec-8685-2079185f8b3d/roslaunch-KI00070-15086.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://KI00070:40915/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_hw/max_range: 4.0
 * /camera/disparity_registered_hw/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/depth_camera_info_url:
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: True
 * /camera/driver/device_id: #1
 * /camera/driver/rgb_camera_info_url:
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /depthimage_to_laserscan/output_frame_id: camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    depthimage_to_laserscan (nodelet/nodelet)
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_hw (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/camera_nodelet_manager-1]: started with pid [15103]
process[camera/driver-2]: started with pid [15104]
process[camera/rgb_debayer-3]: started with pid [15105]
process[camera/rgb_rectify_mono-4]: started with pid [15106]
process[camera/rgb_rectify_color-5]: started with pid [15122]
process[camera/ir_rectify_ir-6]: started with pid [15136]
process[camera/depth_rectify_depth-7]: started with pid [15145]
process[camera/depth_metric_rect-8]: started with pid [15157]
process[camera/depth_metric-9]: started with pid [15179]
process[camera/depth_points-10]: started with pid [15187]
process[camera/register_depth_rgb-11]: started with pid [15199]
process[camera/points_xyzrgb_sw_registered-12]: started with pid [15210]
process[camera/depth_registered_sw_metric_rect-13]: started with pid [15223]
process[camera/depth_registered_rectify_depth-14]: started with pid [15229]
process[camera/points_xyzrgb_hw_registered-15]: started with pid [15248]
process[camera/depth_registered_hw_metric_rect-16]: started with pid [15254]
process[camera/depth_registered_metric-17]: started with pid [15264]
[ INFO] [WallTime: 1654150478.829639824] [node:/camera/camera_nodelet_manager] [func:Loader::Impl::advertiseRosApi]: Initializing nodelet with 4 worker threads.
process[camera/disparity_depth-18]: started with pid [15279]
process[camera/disparity_registered_sw-19]: started with pid [15295]
process[camera/disparity_registered_hw-20]: started with pid [15318]
process[depthimage_to_laserscan-21]: started with pid [15335]
[ INFO] [WallTime: 1654150479.166859103] [node:/camera/camera_nodelet_manager] [func:DriverNodelet::setupDevice]: No devices connected.... waiting for devices to be connected
```

```bash
~ $ rostopic list
/camera/depth_registered/camera_info
/camera/depth_registered/image
/camera/rgb/image_raw
/clicked_point
/initialpose
/map
/map_updates
/move_base/NavfnROS/plan
/move_base_simple/goal
/particlecloud
/rosout
/rosout_agg
/scan
/tf
/tf_static
~ $ rostopic hz /camera/depth_registered/image
subscribed to [/camera/depth_registered/image]
no new messages
no new messages
no new messages
no new messages
```

**To Reproduce**
Steps to reproduce the behavior:
1. 研究室のturtlebot付属のlaptopにKinectを接続
2. mech-userでログイン
3. `roslaunch turtlebot_bringup 3dsensor.launch` を実行

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

```bash
```

kindsenior, 2022-11-13:
自分も学科PCだと同じ症状でした．

どう対処すべきでしょうか？ > @k-okada 
1. 本家のturtlebotにPRする
2. [kindsenior/use-freenect](https://github.com/kindsenior/turtlebot/tree/use-freenect)を使う
3. freenectは何らかの理由で良くない？ので他の解決法を探る？

https://github.com/kindsenior/turtlebot/commit/8a268dadf16cdd6e1248876f43f53b5714886c27
で，デバイスの読み込みに成功して概ね同じトピックが出てそうなところまで確認しました．（演習の全ての課題が動くかどうかは未検証）
が，[turtebot](https://github.com/turtlebot/turtlebot)にPRする必要があるので，演習までにマージされるか（or 間に合うか）微妙．．．

確認項目

1. #305 で同様のことに触れていてるが，turtlebot_bringup/launch/3dsensor.launchを変更してなそうなので，当時はドライバをインストールするだけで動いた？
4. #321 でfreenect のインストールを削除しているのはmelodicのaptがリリースされたからと思われるが，それ以外の理由でfreenectを使わない理由があるか？
5. なぜ本家の[turtebot](https://github.com/turtlebot/turtlebot)が未だにOpenNIを使っているのか．．．

メモ
当初，カーネル依存を疑いlinux-image4.15.0-196-genericとlinux-image-5.4.0-125-genericを試したが，解決できなかった．
（仮に解決できても学科PCでカーネルアップデートはしない？）

k-okada, 2022-11-14:
>    3. なぜ本家のturtebot <https://github.com/turtlebot/turtlebot>

１） PRは送らない理由は何もないので，無条件に送りましょう。
２） マージされる／リリースされる云々は別次元の話なので，それは置いておいて，
https://github.com/jsk-enshu/robot-programming/blob/master/.rosinstall.melodic
を https://github.com/kindsenior/turtlebot/tree/use-freenect
にしましょう。そのさい，なぜこれが必要なのか後で振り返ると分からなくなるので，このPRへのリンクと理由を書いておきましょう。
https://github.com/jsk-ros-pkg/jsk_robot/blob/3a3491973f645268d77db0f9e26fb14b3fb56fbe/jsk_fetch_robot/jsk_fetch.rosinstall.melodic#L4-L7
３） 他の演習との兼ね合いでOpenNI周りのライブラリを入れ替えたのかな。。。。

ソフトウェア第三をまじめにやっていると以下の環境があるはずなので，
その場合はgit pull するとか，wstool update で dだか何かを選ぶとか，書いておいた方がよさそうです。

$ mkdir -p ~/enshu_ws/src # ワークスペースのディレクトリを作成
$ cd ~/enshu_ws/src
$ source /opt/ros/melodic/setup.bash
$ git clone https://github.com/jsk-enshu/robot-programming.git
$ wstool init . # ワークスペースを初期化
# ワークスペースに取得するリポジトリ情報を取得
$ wstool merge robot-programming/.rosinstall.melodic
$ wstool update # ワークスペースにソースコードをダウンロード
$ rosdep update # rosdep ツールの更新
# 依存しているソフトをapt install する．
# Cannot locate rosdep definition ... と
# 表示されるものがあるが無視して良い．
$ rosdep install --from-paths . --ignore-src -y -r
$ cd ..
$ catkin build # 最初の１回は必ず~/enshu_ws ディレクトリで実行すること．

kindsenior, 2022-11-20:
1) 本家へのPR：https://github.com/turtlebot/turtlebot/pull/294
2) 下記で対応しています
https://github.com/jsk-enshu/robot-programming/blob/cc57303668e24c749cb826580bccf783fcfe429d/.rosinstall.melodic#L22-L25

ワークスペースの作成方法について
下記で修正しました
https://sourceforge.net/p/jsk-enshu/code/587/tree//trunk/doc/2022/robot-programming-1-2022.tex?diff=50c99eba34309d09e53b849c:586

kindsenior, 2022-11-28:
@k-okada 

ARVRの演習で学科PC+OpenNI+kinect v1を動かせるのは，openni-dev周りのライブラリを上書きor追加しているようです．

動かし方
[avin2/SensorKinectのv0.93-5.1.2.1](https://github.com/avin2/SensorKinect/tree/v0.93-5.1.2.1)を使ってhttps://github.com/avin2/SensorKinect/blob/v0.93-5.1.2.1/README#L173-L190 に従います．
```
git clone git@github.com:avin2/SensorKinect.git -v v0.93-5.1.2.1
cd <clone directory>/Platform/Linux/CreateRedist
./RedistMaker
cd <clone directory>/Platform/Linux/Redist/Sensor-Bin-Linux-x64-v5.1.2.1 # READMEだとSensor-Bin-Linux-x64-v5.1.2.1ではなくResdistで実行することになっているが，Redistにはinstall.shはない
sudo ./install.sh
```

ARVRの演習では，既にビルドしてあるSensor-Bin-Linux-x64-v5.1.2.1を配布しているようです．
演習ページからDLしたものと上記の方法でビルドしたものにdiffはなくどちらもkinect v1を使えました．

昨年は学科PCではなく自前のPC（おそらく過去にNITEを使ったことがある）で動作確認したため，気づきませんでしたが，ARVRの演習の後にロボットプログラミングの演習を受けると自然とkinectを使えるようになるというからくりな気がします．

確認事項
A) ARVRの担当の先生に手順が同じか確認する？
B) [本家のPrimeSense/Sensor](https://github.com/PrimeSense/Sensor)でない理由（今の所，動かせるリビジョンは未発見）

iory, 2023-11-10:
https://github.com/jsk-enshu/robot-programming/issues/427#issuecomment-1328917199
学科PCでこちらの方法で正常にkinectが動作することを確認しました