jsk-enshu/robot-programming #428: 20221031の資料p17零空間の利用のコードについて

tatsukamijo, 2022-11-02:
ロボットシステム10/31の資料のp17**零空間の利用**の以下のソースコードについて、２点質問があります。
```
(load "irteus/demo/sample-robot-model.l")
(setq r (instance sample-robot :init))
(objects (list r))
(setq t-axis t r-axis t i 0)
(do-until-key
　(setq c (make-cascoords :pos (float-vector (+ (* 100 (sin (/ i 100.0))) 350)
　　　　　　　　　　　　　　　　　　　　　　　　　　　(+ (* 100 (cos (/ i 100.0))) 100) 100)))
　(setq J (send r :calc-jacobian-from-link-list
　　　　　　　　　(send r :link-list (send r :larm :end-coords :parent) (send r :larm :root-link))
　　　　　　　　　:move-target (send r :larm :end-coords)
　　　　　　　　　:rotation-axis r-axis :translation-axis t-axis))
　(setq J# (sr-inverse J))
　(setq I-J#J (m- (unit-matrix (array-dimension J 1)) (m* J# J)))
　(setq dif-pos (send (send r :larm :end-coords) :difference-position c :translation-axis t-axis))
　(setq dif-rot (send (send r :larm :end-coords) :difference-rotation c :rotation-axis r-axis))
　(setq da (v+ (transform J# (concatenate float-vector (calc-dif-with-axis dif-pos t-axis)
　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　(calc-dif-with-axis dif-rot r-axis)))
　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　;;(transform I-J#J (float-vector 0 -0.5 0 0 0 0 0))
　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　　(transform I-J#J (float-vector 0 0.5 0 0 0 0 0))))
　(send r :larm :angle-vector (v+ (send r :larm :angle-vector) da))
　(send *irtviewer* :draw-objects :flush nil)
　(send c :draw-on :flush t)
　(x::window-main-one)
　(incf i)) 
```

1. 上記中ほどの、ロボットに与える関節角度`da`を計算している部分で使われている`calc-dif-with-axis`メソッドの実行内容がわかりません。`t-axis`というのは`translation-axis`と同じものですか？
2. この実装においてはSR-inverseを用いて逆運動学解を直接計算していますが、`:inverse-kinematics`メソッドで`:null-space`を用いても同じことができるということであっていますでしょうか？

k-okada, 2022-11-04:
1
おそくなりました。

1. `t-axis` というのは `translation-axis` は同じで，`calc-dif-with-axis` は `t-axis` を考慮して位置，角度の差分を計算しています。
これは，
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L2827
で定義されています。

2. の理解はその通りです。

tatsukamijo, 2022-11-04:
ご回答いただきありがとうございます。
1について、理解できました。jmanual.pdfを見てわからず詰まっていましたが定義元を直接見に行くべきでした。
2について、:null-space を使ったものも書いて見ようと思います。

closed 2022-11-04 by tatsukamijo