jsk-enshu/robot-programming #429: 1107 "test_object" passed to lookupTransform argument source_frame does not exist.

KajioNaoya, 2022-11-07:
11/7 資料6章の手順に従って実行していったところ、
`(send *l* :lookup-transform "/camera_rgb_optical_frame" "/test_object" (ros::time 0))`
を実行しようとしたところで、以下のエラーが出てしまいました。

[image]

example.launchファイルは Documents/soft3/1107 内に作成し、6章の各コマンドは $ roseus まですべて同ディレクトリ内で実行しました。

可能性のある原因や、原因の調べ方をご教授いただけますと幸いです。

k-okada, 2022-11-08:
`camera_rgb_optical_frame` -> `test_object` のTFはcheckerboard detector画像処理プログラムが出力しますが，そのためにはロボットのカメラにチェッカボードが映っている必要があります．
ロボットを左回転（？）させてカメラにチェッカボード（チェスボード）が映るようにして，
`Checkerboard Detector` という画面で対象物付近に認識結果が表示されているか確認してみてください．

[image]
[image]

KajioNaoya, 2022-11-09:
チェッカーボードをカメラに映した状態で実行したら成功しました。
ありがとうございます。

tomporon, 2022-11-15:
私も同様の問題がありましたが、同じ手順を取ったところうまくいきました。