jsk-enshu/robot-programming #431: launchファイルの使い方について

shryt, 2022-11-15:
launchファイルを使って、gazeboやrqt_guiと一緒に、lispで書いたパブリッシャ・サブスクライバのノードを起動してみたいのですが、nodeタグのpkgやtypeなどの属性をどのように指定したらよいかわからず困っております。第5回6.1の`roslaunch example.launch`の実行はできる状態です。そのため、launchファイルの使い方を教えていただけると幸いです。

ログを見る限り、ROS pathとして`enshu_ws/src`にあるフォルダが認識されているようなので、試しに`enshu_ws/src/test`というフォルダを作り、その中に`1.lisp`というファイルを作成し、'#!/usr/bin/env roseus'を最初に書き、`chmod +x 1.lisp`を実行した状態です(つまり`./1.lisp`で起動できる状態です)。同フォルダに`test.launch`というファイルを作成し、
```
<launch>
    <node pkg="test" type="1.lisp" name="test1" />
</launch>
```
と書いた上で`roslaunch test.launch`を実行してみましたが、やはり認識されない状態です。

以下ログです。
```
mech-user@ki00087:~/enshu_ws/src/test$ roslaunch test.launch 
... logging to /home/mech-user/.ros/log/a8794e5c-6485-11ed-aa42-207918514a2d/roslaunch-ki00087-9469.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ki00087:35799/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    test1 (test/1.lisp)

auto-starting new master
process[master]: started with pid [9496]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a8794e5c-6485-11ed-aa42-207918514a2d
process[rosout-1]: started with pid [9522]
started core service [/rosout]
ERROR: cannot launch node of type [test/1.lisp]: test
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/mech-user/enshu_ws/src/robot-programming/cart_humanoid
ROS path [2]=/home/mech-user/enshu_ws/src/dynamixel_motor/dynamixel_controllers
ROS path [3]=/home/mech-user/enshu_ws/src/dynamixel_motor/dynamixel_driver
ROS path [4]=/home/mech-user/enshu_ws/src/dynamixel_motor/dynamixel_motor
ROS path [5]=/home/mech-user/enshu_ws/src/dynamixel_motor/dynamixel_msgs
ROS path [6]=/home/mech-user/enshu_ws/src/robot-programming/dynamixel_7dof_arm
ROS path [7]=/home/mech-user/enshu_ws/src/dynamixel_motor/dynamixel_tutorials
ROS path [8]=/home/mech-user/enshu_ws/src/dynamixel_urdf
ROS path [9]=/home/mech-user/enshu_ws/src/kobuki/kobuki
ROS path [10]=/home/mech-user/enshu_ws/src/kobuki/kobuki_auto_docking
ROS path [11]=/home/mech-user/enshu_ws/src/kobuki/kobuki_bumper2pc
ROS path [12]=/home/mech-user/enshu_ws/src/kobuki/kobuki_capabilities
ROS path [13]=/home/mech-user/enshu_ws/src/kobuki/kobuki_controller_tutorial
ROS path [14]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_dashboard
ROS path [15]=/home/mech-user/enshu_ws/src/kobuki/kobuki_description
ROS path [16]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_desktop
ROS path [17]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_gazebo
ROS path [18]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_gazebo_plugins
ROS path [19]=/home/mech-user/enshu_ws/src/kobuki/kobuki_keyop
ROS path [20]=/home/mech-user/enshu_ws/src/kobuki/kobuki_random_walker
ROS path [21]=/home/mech-user/enshu_ws/src/kobuki/kobuki_rapps
ROS path [22]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_rviz_launchers
ROS path [23]=/home/mech-user/enshu_ws/src/kobuki/kobuki_safety_controller
ROS path [24]=/home/mech-user/enshu_ws/src/kobuki/kobuki_node
ROS path [25]=/home/mech-user/enshu_ws/src/kobuki/kobuki_testsuite
ROS path [26]=/home/mech-user/enshu_ws/src/kobuki_desktop/kobuki_qtestsuite
ROS path [27]=/home/mech-user/enshu_ws/src/robot-programming/mechatrobot
ROS path [28]=/home/mech-user/enshu_ws/src/turtlebot/turtlebot
ROS path [29]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_actions
ROS path [30]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_apps
ROS path [31]=/home/mech-user/enshu_ws/src/turtlebot/turtlebot_bringup
ROS path [32]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_calibration
ROS path [33]=/home/mech-user/enshu_ws/src/turtlebot/turtlebot_capabilities
ROS path [34]=/home/mech-user/enshu_ws/src/turtlebot_interactions/turtlebot_dashboard
ROS path [35]=/home/mech-user/enshu_ws/src/turtlebot/turtlebot_description
ROS path [36]=/home/mech-user/enshu_ws/src/turtlebot_simulator/turtlebot_gazebo
ROS path [37]=/home/mech-user/enshu_ws/src/turtlebot_interactions/turtlebot_interactions
ROS path [38]=/home/mech-user/enshu_ws/src/turtlebot_interactions/turtlebot_interactive_markers
ROS path [39]=/home/mech-user/enshu_ws/src/turtlebot_msgs
ROS path [40]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_follower
ROS path [41]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_navigation
ROS path [42]=/home/mech-user/enshu_ws/src/turtlebot_apps/turtlebot_rapps
ROS path [43]=/home/mech-user/enshu_ws/src/turtlebot_interactions/turtlebot_rviz_launchers
ROS path [44]=/home/mech-user/enshu_ws/src/turtlebot_simulator/turtlebot_simulator
ROS path [45]=/home/mech-user/enshu_ws/src/turtlebot_simulator/turtlebot_stage
ROS path [46]=/home/mech-user/enshu_ws/src/turtlebot_simulator/turtlebot_stdr
ROS path [47]=/home/mech-user/enshu_ws/src/turtlebot/turtlebot_teleop
ROS path [48]=/home/mech-user/enshu_ws/src/robot-programming/turtleboteus
ROS path [49]=/home/mech-user/enshu_ws/src/robot-programming/dxl_armed_turtlebot
ROS path [50]=/home/mech-user/enshu_ws/src/robot-programming/daisya_euslisp_tutorials
ROS path [51]=/opt/ros/melodic/share
```

k-okada, 2022-11-15:
```    <node pkg="test" type="1.lisp" name="test1" />```は `test`というパッケージ内の`1.lisp`という実行ファイルを実行します。
`test`というディレクトリがあれば，`test`というパッケージというわけではなくて，`test`というディレクトリがあったうえで，いくつか条件があります。おおざっぱには，`package.xml` と `CMakeLists.txt`があればよいのですが，その中に書くべき内容はいろいろルールがあるので，一回`test`ディレクトリを消して，
`catkin_create_pkg test roseus`
として作成しなおして，そこに`1.lisp`を追加して，さらに`catkin build` して`source devel/setup.bash`すると`rospack find test` でパッケージが見つかれば，`launch`ファイルも動くと思います。

shryt, 2022-11-15:
無事にroslaunchで実行することができました。ありがとうございました。ちなみにここでの質問はCloseしないほうが良いでしょうか。

k-okada, 2022-11-15:
他にも同じところで困っている人がいるとおもうので、とりあえずこのままでokです

shryt, 2022-11-15:
わかりました。ありがとうございました。